knurling-rs/defmt #132: Undefined reference to `_defmt_acquire` `_defmt_release`

mattico, 2020-08-25:
My program links to a C library which depends on libc, so I'm linking with GCC like so:

```toml
[build]
target = "thumbv7em-none-eabihf"
rustflags = [
  "-C", "inline-threshold=225",
  "-C", "linker=arm-none-eabi-gcc",
  "-C", "link-arg=-Wl,-Tlink.x",
  "-C", "link-arg=-Wl,-Tdefmt.x",
  "-C", "link-arg=-nostartfiles",
  "-C", "link-arg=-specs=nano.specs",
  "-C", "link-arg=-lc",
  "-C", "link-arg=-mcpu=cortex-m7",
  "-C", "link-arg=-mfpu=fpv5-d16",
  "-C", "link-arg=-mfloat-abi=hard",
  "-C", "link-arg=-mthumb",
]
```

I'm using `defmt-rtt` from git and `use defmt_rtt as _;`. Linking fails with the following error:
```
note: c:/program files (x86)/gnu arm embedded toolchain/9 2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\matth\Code\rs_firmware\target\thumbv7em-none-eabihf\debug\deps\libdefmt-0558b8a194b288d7.rlib(defmt-0558b8a194b288d7.defmt.bhmz7gi0-cgu.5.rcgu.o): in function `defmt::export::acquire':   
          C:\Users\matth\.cargo\git\checkouts\defmt-7f5b74b4e6ff55d4\5a1047b\src/export.rs:49: undefined reference to `_defmt_acquire'
          c:/program files (x86)/gnu arm embedded toolchain/9 2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\matth\Code\rs_firmware\target\thumbv7em-none-eabihf\debug\deps\libdefmt-0558b8a194b288d7.rlib(defmt-0558b8a194b288d7.defmt.bhmz7gi0-cgu.5.rcgu.o): in function `defmt::export::release':   
          C:\Users\matth\.cargo\git\checkouts\defmt-7f5b74b4e6ff55d4\5a1047b\src/export.rs:60: undefined reference to `_defmt_release'
          collect2.exe: error: ld returned 1 exit status
```

Using readelf on defmt_rtt reveals:
```
   241: 00000001    42 FUNC    GLOBAL DEFAULT   18 _defmt_acquire
   242: 00000001    66 FUNC    GLOBAL DEFAULT   22 _defmt_release
```
Using readelf on defmt reveals:
```
    75: 00000000     0 NOTYPE  GLOBAL DEFAULT  UND _defmt_acquire
    76: 00000000     0 NOTYPE  GLOBAL DEFAULT  UND _defmt_release
    77: 00000000     0 NOTYPE  GLOBAL DEFAULT  UND _defmt_timestamp
```

Adding `-u` args to not discard these symbols did not help.
```diff
+  "-C", "link-arg='-u _defmt_acquire'",
+  "-C", "link-arg='-u _defmt_release'",
```

Adding `#[no_mangle]` to the extern blocks in export.rs did not help.

I'd appreciate any ideas!

jamesmunns, 2020-08-26:
It looks like you are missing a [`#[global_logger]`](https://defmt.ferrous-systems.com/global-logger.html) implementation. `use defmt_rtt as _;` should handle that, so I'm not sure why it isn't getting linked in

japaric, 2020-08-26:
@mattico please try PR #134 and make sure you have `#[global_logger]` in the dependency graph (`use defmt_rtt as _;` at the top should take care of that)

PR #134 (2020-08-26): linker-script: exhaustively search for defmt symbols
should fix #132
closed 2020-08-26 by japaric