knurling-rs/defmt #135: Error: one or more of debug, error, info, trace, warn ranges overlap

andresv, 2020-08-26:
I tried to setup `defmt` for an old project.
I took `lib.rs` straight from https://github.com/knurling-rs/app-template/blob/main/src/lib.rs and modified `config.toml` according to this: https://github.com/knurling-rs/app-template/blob/main/.cargo/config.toml

My `Cargo.toml` looks more or less like this:
```toml
...
name = "um8"
...

[dependencies.defmt]
git = "https://github.com/knurling-rs/defmt"
branch = "main"

[dependencies.defmt-rtt]
git = "https://github.com/knurling-rs/defmt"
branch = "main"

[dependencies]
panic-halt = "0.2"
cortex-m = "0.6.3"
cortex-m-rt = "0.6.12"
embedded-hal = "0.2"
cortex-m-rtic = "0.5"
...

[[bin]]
name = "um8"
test = false
bench = false

[lib]
name = "defmtlib"
path = "src/lib.rs"
test = false
bench = false

[features]
# set logging levels here
default = [
    # enable logs of this crate (`app`) at the TRACE level
    "defmt-trace",
    #"some_module/defmt-debug",
]
# do NOT modify these features
defmt-default = []
defmt-trace = []
defmt-debug = []
defmt-info = []
defmt-warn = []
defmt-error = []

[profile.dev]
opt-level = 1
codegen-units = 16
debug = 1
lto = 'fat'

[profile.release]
opt-level = 2 # optimize for speed
codegen-units = 1 # better optimizations
debug = 1 # symbols are nice and they don't increase the size on Flash
lto = 'fat'
```

Notice that `src/lib.rs` is named to `defmtlib`.
`main.rs` looks like this:
```rust
...
use defmtlib as _; // src/lib.rs
...
#[init(spawn = [blinky])]
    fn init(mut ctx: init::Context) -> init::LateResources {
        defmt::info!("Hello, world!");
        //defmt::error!("Hello, world!");
```

Logging works with `error!` but not with `info!`:
```
Error: one or more of debug, error, info, trace, warn ranges overlap
```

japaric, 2020-08-26:
uh oh, that error message was introduced in #130 so the logic may contain a bug.

jonas-schievink, 2020-08-26:
So I think the only problem was that empty ranges were not accounted for, which should be fixed by https://github.com/knurling-rs/defmt/pull/139. Would be great if you could try that out!

andresv, 2020-08-26:
```
cargo install \                                         
--git https://github.com/knurling-rs/probe-run \
--branch main \
--features defmt
```
says:
```
    Updating git repository `https://github.com/knurling-rs/probe-run`
     Ignored package `probe-run v0.1.3 (https://github.com/knurling-rs/probe-run?branch=main#7ce94c22)` is already installed, use --force to override
```
However I updated `defmt` and now firmware and tools do not match.

jonas-schievink, 2020-08-26:
Yeah, unfortunately you have to install the exact same git commit for both, so try passing `--force` to `cargo install` as well.

andresv, 2020-08-26:
In firmware I can do this:
```
[dependencies]
defmt = { git = "https://github.com/knurling-rs/defmt", branch = "main" }
defmt-rtt = { git = "https://github.com/knurling-rs/defmt", branch = "main" }
```

However `probe-run` is in other repo, so there isn't same git hash in there?

andresv, 2020-08-27:
Alright I can confirm that it works.

I cloned probe-run and changed:
```
decoder = { git = "https://github.com/knurling-rs/defmt", branch = "overlap-fix", optional = true }
elf2table = { git = "https://github.com/knurling-rs/defmt", branch = "overlap-fix", optional = true
```
And then built with:
`cargo install --path "probe-run" --features defmt`

jonas-schievink, 2020-08-27:
Great, thanks for testing!

closed 2020-08-27 by jonas-schievink