knurling-rs/defmt #140: probe-rs loses connection with stm32 if stm performs flash write

andresv, 2020-08-27:
`defmt` version: `overlap-fix#0ceade69`

I am using stm32g081. In my app first couple of flash pages are allocated to bootloader and configuration.
When using `defmt` and at the same time stm32 tries to write to configuration page, communication with the probe will be lost:

```
0.000000 INFO app info: v0.9.2 hw: 8 ts: 0
Error: A core architecture specific error occured

Caused by:
    0: Failed to read register DRW at address 0x0000000c because: An error specific to a probe type occured
    1: An error specific to a probe type occured
    2: Command failed with status SwdDpWait
```

japaric, 2020-08-27:
Does this issue also occur with `rtt-target` + `cargo-embed`? If yes, then this could be an upstream issue and should be reported to the probe-rs issue tracker.

I'm only familiar with the low level details of CMSIS-DAP + SWD (I'm not sure which probe you are using. st-link?) but RTT running in parallel to device-side Flash operations seems like it should be allowed at the architecture (ARM Cortex-M) level. Now if the stm32 chips in particular disallow this I wouldn't know.

Urhengulas, 2021-02-28:
Hi @andresv,
Does this problem persist when using the `v0.2.0` of `probe-run` and `defmt`?

MathiasKoch, 2021-03-01:
I am currently running into this with v0.1.x. I'll try to upgrade my stack to v0.2.0, and check if it still persists.

I would expect so though, as i had a chat with the probe.rs guys, who told me that it might be an ST-link v2 issue, that cannot be resolved by them.

MathiasKoch, 2021-03-01:
Confirmed, this is still an issue with 
```
probe-run --version
0.2.1
supported defmt version: 0.2
```

and `defmt = "0.2.0"`

Urhengulas, 2021-03-01:
Okay. Good to know it still persists.

> I would expect so though, as i had a chat with the probe.rs guys, who told me that it might be an ST-link v2 issue, that cannot be resolved by them.

@MathiasKoch: Could you please quickly summarize what the probe-rs people told you? And if you think this could be resolved on our side?

MathiasKoch, 2021-03-01:
**MathiasKoch**
> ```
>  (HOST) DEBUG Scanning at exact address: 0x20000000
>└─ probe_rs_rtt::rtt @ /home/mathias/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-rtt-0.10.1/src/rtt.rs:163
> (HOST) DEBUG Read mem 8 bit, address=20000000, length=24
>└─ probe_rs::probe::stlink @ /home/mathias/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs->0.10.1/src/probe/stlink/mod.rs:859
> (HOST) DEBUG Read mem 8 bit, address=20000018, length=24
>└─ probe_rs::probe::stlink @ /home/mathias/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs->0.10.1/src/probe/stlink/mod.rs:859
>  (HOST) DEBUG Read mem 8 bit, address=08008dfc, length=64
>└─ probe_rs::probe::stlink @ /home/mathias/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs->0.10.1/src/probe/stlink/mod.rs:859
>  (HOST) WARN  send_jtag_command 242 got SwdDpWait, retrying
>└─ probe_rs::probe::stlink @ /home/mathias/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs->0.10.1/src/probe/stlink/mod.rs:700
>  (HOST) WARN  send_jtag_command 242 got SwdDpWait, retrying
>└─ probe_rs::probe::stlink @ /home/mathias/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs->0.10.1/src/probe/stlink/mod.rs:700
>  (HOST) WARN  send_jtag_command 242 got SwdDpWait, retrying
>└─ probe_rs::probe::stlink @ /home/mathias/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs->0.10.1/src/probe/stlink/mod.rs:700
>  (HOST) WARN  send_jtag_command 242 got SwdDpWait, retrying
>└─ probe_rs::probe::stlink @ /home/mathias/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs->0.10.1/src/probe/stlink/mod.rs:700
>  (HOST) WARN  send_jtag_command 242 got SwdDpWait, retrying
>└─ probe_rs::probe::stlink @ /home/mathias/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs->0.10.1/src/probe/stlink/mod.rs:700
>  (HOST) WARN  send_jtag_command 242 got SwdDpWait, retrying
>└─ probe_rs::probe::stlink @ /home/mathias/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs->0.10.1/src/probe/stlink/mod.rs:700
>  (HOST) WARN  send_jtag_command 242 got SwdDpWait, retrying
>└─ probe_rs::probe::stlink @ /home/mathias/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs->0.10.1/src/probe/stlink/mod.rs:700
>  (HOST) WARN  send_jtag_command 242 got SwdDpWait, retrying
>└─ probe_rs::probe::stlink @ /home/mathias/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs->0.10.1/src/probe/stlink/mod.rs:700
>  (HOST) WARN  send_jtag_command 242 got SwdDpWait, retrying
>└─ probe_rs::probe::stlink @ /home/mathias/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs->0.10.1/src/probe/stlink/mod.rs:700
>  (HOST) WARN  send_jtag_command 242 got SwdDpWait, retrying
>└─ probe_rs::probe::stlink @ /home/mathias/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs->0.10.1/src/probe/stlink/mod.rs:700
> (HOST) WARN  send_jtag_command 242 got SwdDpWait, retrying
>└─ probe_rs::probe::stlink @ /home/mathias/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs->0.10.1/src/probe/stlink/mod.rs:700
>  (HOST) WARN  send_jtag_command 242 got SwdDpWait, retrying
>└─ probe_rs::probe::stlink @ /home/mathias/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs->0.10.1/src/probe/stlink/mod.rs:700
>  (HOST) WARN  send_jtag_command 242 got SwdDpWait, retrying
>└─ probe_rs::probe::stlink @ /home/mathias/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs->0.10.1/src/probe/stlink/mod.rs:700
>  (HOST) WARN  too many retries, giving up
>└─ probe_rs::probe::stlink @ /home/mathias/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs->0.10.1/src/probe/stlink/mod.rs:715
>  (HOST) DEBUG Read mem 32 bit, address=e0002000, length=4
>└─ probe_rs::probe::stlink @ /home/mathias/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs->0.10.1/src/probe/stlink/mod.rs:793
>  (HOST) DEBUG Current device mode: Jtag
>└─ probe_rs::probe::stlink @ /home/mathias/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs->0.10.1/src/probe/stlink/mod.rs:402
>Error: Error communicating with probe: An error with the usage of the probe occured
>
>Caused by:
>    0: An error with the usage of the probe occured
>    1: An error specific to a probe type occured
>    2: Command failed with status SwdDpWait
>```

**dirbaio**

> https://github.com/probe-rs/probe-rs/pull/370/commits/e889c664ca4dda039c153b27aad51c1f125065d0
> fixed in 0.10

**yatekii**
> yep, as you can see, new probe-rs retries :)
> but apparently not long enough

**dirbaio**
>still fails? O_o
>@yatekii:matrix.org
>yeah, see log =)

**MathiasKoch**
>for attempt in 0..13 { why 13 ? :p
>@yatekii:matrix.org

**MathiasKoch**
>That changed it a bit
[... 254 lines cut ...]
**yatekii**
>ah, I didn't mean that. I mean we could with some other command.

**dirbaio**
>halting the core? but probe-rs does that already right?

**dirbaio**
>I thought the ABORT aborts ap/dp transaction, not mcu transactions
>ABORT does not really help if the CPU locks forever. so yeah ^^

**dirbaio**
>halting the core? but probe-rs does that already right?
>yeah but at points where specifically instructed to do so.
>not just randomly on some timeout ^^
>or rather: halting is okay
>but not altering the control flow at some random point
>imagine trying to use flash from your FW and the debugger always cancels the transfer because it takes a split second too long :D
>I would become mad nuts :D

**dirbaio**
>yeah it sucks 💩
>I'd try the ABORT in the retry loop though, it's something we haven't tried and the ARM docs explicitly say to do that

**yatekii**
>yes, ofc, I agree it's worth a try!

Urhengulas, 2021-03-01:
So the consensus is to try the `ABORT`, but more generally just not use this debugger?

Yatekii, 2021-03-01:
Basically, the ST-Links are unpredictable because they do not have a documented interface. The issues we have in probe-rs are issues with other libs sometimes too and sometimes they handle things better. It's something that will get better over time with more time thrown onto the problem. As long as probe-rs is unpaid work, I doubt anyone is willing to invest substantial amounts of time for the remaining 5% of issues when one can use a properly documented probe and invest that time into solving actually unsolved problems.

We can try the ABORT fix; it should not be hard. We'll see.

Urhengulas, 2021-03-01:
Thanks for your answer @Yatekii! To help people stumbling upon this issue, could you maybe tell what probe would you recommend from a user experience and developer experience perspective?

Else I'd close this here and defer to `probe-rs`. Please send a small update in case there is progress!

closed 2021-03-01 by Urhengulas

Yatekii, 2021-03-01:
So, hardware wise, the ST-Link is not bad. The same goes for the J-Link. But they both have a closed source interface which was partially reverse engineered.
The third probe type of the ones we support at the moment is the CMSIS-DAP. There is v1 and v2. v1 is limited in speed by USB HID (~20kBit actual transfer speed). v2 is actually really fast and can, if the chip allows it go to a few hundred kbit/s. Then it hits the limit of SWD and the on chip flash.
The cool thing about CMSIS-DAP is that it is open source. So we decide about every detail of the dependency chain basically :)
The problem with CMSIS-DAP is that there is more or less only "devkit-probes" which kind of require you to flash the firmware yourself and are a barebones PCB. For most people this is not an issue.

At this point I'd like tho throw in the hs-probe: https://docs.google.com/forms/d/1id0tnkM3Ail0QE4CNcp19bbWmHBvWrO-tu5cQtQ8Azc/ into the ring.
We are currently finalizing the orders for the first batch which got only held up by Chinese NY.
The hs-probe features a 100% Rust firmware with an implementation of CMSIS-DAP v1 and v2. It can also be used to program FPGAs. Currently we make no money of this (we have a little bit of safety margin, in just in case which would go into the probe-rs fund), so I hope it's ok to advertise this.

Urhengulas, 2021-03-01:
Thank you for your thoughts on the probes!

And `hs-probe` seems like an awesome project 🤯  Props for this 👏🏾

Yatekii, 2021-03-01:
All props to @korken89, @disasm and @adamgreig :)