knurling-rs/defmt #142: Support formatting &[u8] as ASCII strings

birkenfeld, 2020-08-27:
I'm often working with byte slices that have string (or mixed) contents, however it's not worth doing the `str::from_utf8` dance *just* for logging purposes.

So it would be very convenient to have a format specifier that outputs the slice contents as ASCII, with non-printable and non-ASCII bytes either replaced or escaped.

I'm willing to implement it in case it's deemed acceptable :)

jonas-schievink, 2020-08-27:
This seems useful to have, yeah. There's an interesting design question though: This is, for all intents and purposes, still a `{:[u8]}`, just with a display hint, so it is very similar to a hypothetical `{:u8:x}` modifier that prints an integer in hexadecimal (syntax tbd, we don't support this at the moment). I'd very much like to see the hex modifier at some point, so maybe it makes sense to make this an `:ascii` modifier, like `{:[u8]:ascii}`?

japaric, 2020-08-27:
For the same rationale as https://github.com/knurling-rs/defmt/issues/145#issuecomment-681887216 I would like to see this handled in the frontend.

Now if we are talking about formatting a `AsciiStr` type (which sadly is not in core) I would be in favor of adding a new formatting paremeter for that type. Though I don't think that's what birkenfeld is after.

birkenfeld, 2020-08-27:
@japaric I see the point, especially for integers it's nice to get different reprs automatically, without the firmware specifying it explicitly.

Still, since I (and many others) will be happy with the simple stdout logging that `probe-run` provides, it'd be nice to get a commitment that it will support this, and not only some (hypothetical) web/GUI frontend :)

japaric, 2020-11-24:
this feature is being considered in the 'display hints' RFC (see #270). let's centralize the conversion in that thread. I'm going to close this thread to avoid having parallel on-going conversations

closed 2020-11-24 by japaric