knurling-rs/defmt #145: Hex formating is missing {:x}

andresv, 2020-08-27:
I actually saw that `{:x}` was announced in https://ferrous-systems.com/blog/defmt/. However it seems to be actually missing.
I couldn't find it from here: https://github.com/knurling-rs/defmt/blob/main/parser/src/lib.rs.

japaric, 2020-08-27:
> I saw that {:x} was announced in https://ferrous-systems.com/blog/defmt/. 

the `{:x}` there is making reference to `core::fmt::Hex`. `defmt` has a single "formatting" trait (`Format`) so `{:x}`, `{:b}`, etc. are not planned. 

The rationale for a single trait is that since formatting is done on the host, it should be the host side who decides how to format the string and not the device. Interactive web / GUI front ends can provide a mechanism (e.g. click the number) to change how the information is presented; that way you don't have to re-build, re-flash and re-run the program to change the representation. That however is harder to do with the current non-interactive terminal output so we have considered printing different representations at once (e.g. `{16,0x10,0b10000}` instead of `16`) but that "bloats" the output.

andresv, 2020-08-27:
Got it.

AndWass, 2020-08-28:
> the {:x} there is making reference to core::fmt::Hex. defmt has a single "formatting" trait (Format) so {:x}, {:b}, etc. are not planned.
> The rationale for a single trait is that since formatting is [...]

But allowing the target side to provide formatting hints (via `{:x}` and `{:b}`) doesn't stop interactive web / GUI frontends from providing the functionality to change how information is presented anyway. And isn't it the case that the person who most often will look at the output is also the person who will be writing the code for the target? So for that person it does provide immediate QoL-improvements for the current non-interactive terminal output, especially since the need for currently non-existing tools diminishes.

Dirbaio, 2020-10-19:
+1000 to this. I've been using `defmt` in my projects since it came out, and not being able to print stuff in hex is the only big usability issue I've ran into. And it's quite a big one IMO. Hex is especially important for embedded dev. 

We really need to do this. Say you're printing temperatures in Celsius and pointers / register values. You want the temperatures in decimal and the pointers in hex. Hex temperatures and decimal pointers are useless.

Even if defmt has interactive gui/web frontends that allow you to change the format on the fly, specifying hex/dec is still necessary for the tool to know what to show by default. For example, if the GUI tool showed everything as decimal, you'll always be right-clicking decimal pointers to change them to hex, which will be extremely tedious.

So, here's a rough proposal: We separate the concepts of "type" and "format".

- "type" is the current data type: `u8`, `u32`, `[u8]`, `?`, `str`, `istr`, etc.
- "format" is info on how the data type should be printed. This could be stuff like:
   - base: `d`, `x`, `b`
   - digit count after/before decimal point for floats
   - max/min widths, zero-padding?
   - something I miss even in std `fmt` is being able to print a `[u8]` as an hex string, like `ff49b1`, not like `[0xff, 0x49, 0xb1]`.
   - it'd also be quite useful to be able to print a `[u8]` as an hex dump (16 bytes per line, hex + ascii), like this:
```
 f9 05 ef 3f 41 54 2b 43 47 44 43 4f 4e 54 3d 31  ...?AT+CGDCONT=1
 2c 22 49 50 22 2c 22 6d 32 6d 2e 6e 61 74 2e 65  ,"IP","m2m.nat.e
 73 22 0d 98 f9                                   s"...
```

The format is only used by the host. The target only uses the type.

The target takes the data and encodes it into the stream, with the wire format depending only on the type, exactly how it's done currently. There's still only one `Format` trait, same as currently. There's no extra target-side cost introduced by implementing this.

The host uses the types to decode the stream, and then uses the formats to print the logs. Non-interactive frontends would simply use the chosen format. Interactive frontends would use it as a default, still allowing the user to right-click to change it.

The syntax could be an optional `:format` added at the end. For example:

```rust
info!("{:u32}",      255);     // 255 (default is decimal)
info!("{:u32:d}",    255);     // 255
info!("{:u32:x}",    255);     // ff
info!("{:u32:08x}",  255);     // 000000ff
info!("{:u32:08X}",  255);     // 000000FF
info!("{:u32:b}",    255);     // 11111111
info!("{:u32:032b}", 255);     // 00000000000000000000000011111111
info!("{:[u8]}",     [2,128]); // [2, 128]
info!("{:[u8]:x}",   [2,128]); // [2, 80]
info!("{:[u8]:02x}", [2,128]); // [02, 80]
info!("{:[u8]:02x}", [2,128]); // [02, 80]
info!("{:?}", Foo{ temp: 12, pressure: 34 } );   // { temp: 12, pressure: 34 }
info!("{:?:x}", Foo{ temp: 12, pressure: 34 } ); // { temp: c, pressure: 22 }

```

twitchyliquid64, 2020-11-22:
Hey @jonas-schievink & @japaric, are there any changes to the 'formatting hints' approach or syntax?

Ive watched a few conversations in matrix on this, but I didn't observe us arriving at any different plan.

If the formatting hints approach seems pretty stable, I can polish off the PR and add the remaining formats postulated in https://github.com/knurling-rs/defmt/issues/145#issuecomment-712235944.

japaric, 2020-11-24:
this feature is being considered in the 'display hints' RFC (see #270). let's centralize the conversion in that thread. I'm going to close this thread to avoid having parallel on-going conversations

> I can polish off the PR

note that the feature requires breaking changes (that PR breaks compatibility with existing probe-run versions) so a PR implementing the feature will not be merged until it's clear what's else will go in a 0.2.0 release

closed 2020-11-24 by japaric