knurling-rs/defmt #156: Error: BUG in DWARF variable filter: index collision

MathiasKoch, 2020-09-03:
Running `probe-run` with `RUST_LOG=trace` gives me:
```
[2020-09-03T10:23:17Z DEBUG probe_rs::config::registry] Searching registry for chip with name STM32L475VGTx
[2020-09-03T10:23:17Z DEBUG probe_run] RAM region: 0x20000000-0x20017FFF
Error: BUG in DWARF variable filter: index collision
```

An ELF file reproducing can be found here: https://www.dropbox.com/s/5awbjmyczagrp75/initial?dl=0

compiled with `defmt` commit #cf8a8e8328cd371c2d3ef3198ffb62c219e4dd6e

jonas-schievink, 2020-09-03:
Added some more info:

> Error: BUG in DWARF variable filter: index collision for addr 0x00000001 (old = Location { file: "/home/mathias/.cargo/git/checkouts/mqttrust-76b1690cafd0c799/879e75a/src/state.rs", line: 354 }, new = Location { file: "/home/mathias/.cargo/git/checkouts/ublox-cellular-rs-b58122bcae038225/1be5aeb/ublox-cellular/src/modules/socket.rs", line: 179 })

Not sure if address 1 is special in some way in DWARF but neither location looks correct (address 1 is the initial stack pointer value, not code).

MathiasKoch, 2020-09-03:
One is this line: https://github.com/BlackbirdHQ/mqttrust/blob/879e75ab6ef003616e367781f38e53171e89916d/src/state.rs#L354

Other is: https://github.com/BlackbirdHQ/ublox-cellular-rs/blob/1be5aeb8d5c24cdcd404631af950f1e29c94be29/ublox-cellular/src/modules/socket.rs#L179

japaric, 2020-09-03:
> Not sure if address 1 is special in some way in DWARF

those addresses *should* be the addresses that appear in the output of `nm`; they are the "string index". Looking at the provided ELF file (`initial`) "{:str}" has address `8`, not `1`; the other string ( "WrongSocketType") is not present in the ELF (so it got GC-ed).

@MathiasKoch was this compiled *without* LTO? could you try enabling LTO and see if that "fixes" the problem?

I think it could be that when not using LTO, LLVM produces several DWARF "object file" things where string indices are relative rather than absolute (like it would be the case with relocatable object files `.o`). That would certainly be annoying to fix.

MathiasKoch, 2020-09-03:
It was compiled with 
```
[profile.release]
opt-level = "z"
codegen-units = 1
debug = true
lto = 'fat'
debug-assertions = false

[profile.release.build-override]
codegen-units = 8
debug = false
debug-assertions = false
opt-level = 0
overflow-checks = false

```

MathiasKoch, 2020-09-03:
Not sure if it should be allowed to log in loops like:
```rust
for topic in &unsubscribe_request.topics {
    defmt::trace!("{:str}", topic.as_str());
}
```
?

> you compiled it with cargo build --release then? in that case, the problem could be caused by using defmt from multiple crates rather than by not enabling LTO.

Yep.

And you are correct, defmt is from multiple crates, but shouldn't that be okay?

japaric, 2020-09-03:
> Not sure if it should be allowed to log in loops like:

that's an allowed operation

> And you are correct, defmt is from multiple crates, but shouldn't that be okay?

that *should* work fine but it seems an assumption I made in the implementation doesn't hold in that case (multiple crates) so that should be fixed

MathiasKoch, 2020-09-03:
I can test it..   just a second

japaric, 2020-09-03:
`#[used]` is not required in this case because the control flow is taking the address to the static (you need `used` when the compiler "cannot" see the use of the symbol). We do want unused symbols to be GC-ed (that keeps the string indices small, e.g. below 16 bits)

MathiasKoch, 2020-09-03:
Still an issue with `#b2bef93`

japaric, 2020-09-03:
I analyzed the provided ELF file with `gimli`'s `dwarfdump` tool and it seems "live" symbols (symbols that are kept after linking) are assigned absolute addresses in the DWARF (which is what we want). The issue seems to be that "ghost" symbols (symbols removed by the linker but whose info remains in DWARF) are being assigned addresses "0" and "1".

The assumption was that they would only be assigned address "0" so we made string indices start at "1" to avoid collisions. That assumption was wrong so the tool is seeing "ghost" symbols collide with live symbols. I think instead of assuming that "0" means "ghost symbol" we should look for some other way to discard "ghost" symbols when we analyze the DWARF info.

MathiasKoch, 2020-09-07:
Is there any quick workaround i can use until this is fixed?

japaric, 2020-09-09:
@MathiasKoch #169 and knurling-rs/probe-run#61 should fix *this* issue but if you hit #159 you will still not get location info

PR #169 (2020-09-09): elf2table: discard loc-info from linker GC-ed symbols
should fix #156
closed 2020-09-09 by japaric