knurling-rs/defmt #165: How to use stringify with defmt ?

Dirbaio, 2020-09-06:
I'm trying to do something like this bit of code

```rust

    macro_rules! my_assert {
        ($e:expr) => {
            if !$e {
                defmt::error!("Assertion failed: {:istr}", intern!(stringify!($e)));
            }
        };
    }

    my_assert!(1 == 2);
```

The goal is that my_assert would print `Assertion failed: 1 == 2`. However this fails with this error:

```
error: expected literal
  --> xx/src/lib.rs:82:68
   |
82 |                 defmt::error!("Assertion failed: {:istr}", intern!(stringify!($e)));
   |                                                                    ^^^^^^^^^
...
87 |     my_assert!(1 == 2);
   |     ------------------- in this macro invocation
   |
```

A similar thing happens when trying to use `stringify!` in the first argument of print/error/etc:

```rust
    macro_rules! my_assert {
        ($e:expr) => {
            if !$e {
                defmt::error!(stringify!(Assertion failed: $e));
            }
        };
    }
```

This is because rust expands outer macros first, inner macros last. When it's time to expand `intern!` it finds something that's not a string literal and fails.

Maybe the defmt macros should detect stringify! in the input and expand it?

jonas-schievink, 2020-09-07:
This would require eagerly expanding `stringify!(...)`, which is not currently possible in Rust.

japaric, 2020-09-07:
The first argument of the logging macros needs to be a string literal; it cannot be a macro like `stringify!` because macros don't eagerly expand.
To implement this kind of `assert!` you would need to write a procedural macro.

jonas-schievink, 2020-11-10:
The `intern!` docs should be adjusted to explain that it takes precisely one string literal as argument. The rest is not defmt-specific and works the same way in the `log` crate.

jonas-schievink, 2020-11-10:
Actually it does already say that, so closing this issue. For defmt-specific `assert!` macros, see #136.

closed 2020-11-10 by jonas-schievink