knurling-rs/defmt #172: rust-lld: error: undefined symbol: _defmt_acquire

chenyanming, 2020-09-17:
I can not run `cargo build`, would you help check?

```
error: linking with `rust-lld` failed: exit code: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "--eh-frame-hdr" "-L" "/Users/chandamon/.rustup/toolchains/nightly-x86_64-apple-darwin/lib/rustlib/thumbv7em-none-eabihf/lib" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/ok-00c3d462160d63f2.ok.e22vskm7-cgu.0.rcgu.o" "-o" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/ok-00c3d462160d63f2" "--gc-sections" "-L" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps" "-L" "/Users/chandamon/Documents/freshup/atom/ok/target/debug/deps" "-L" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/build/ok-fe8fce64b33a35b5/out" "-L" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/build/cortex-m-51920042ee9b2630/out" "-L" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/build/cortex-m-rt-48479f92d79eb74e/out" "-L" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/build/cortex-m-semihosting-77cade760834610c/out" "-L" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/build/defmt-4430fad72716f0f8/out" "-L" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/build/stm32f4-14cfb699ee37a4ad/out" "-L" "/Users/chandamon/.rustup/toolchains/nightly-x86_64-apple-darwin/lib/rustlib/thumbv7em-none-eabihf/lib" "-Bstatic" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libdefmt-36442d255a9c07bc.rlib" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libdefmt_common-8a53ad6740ef6f9c.rlib" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libheapless-5c168228bd552c76.rlib" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libhash32-f002e42d4cde7860.rlib" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libbyteorder-f86e7ae79197b381.rlib" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libstm32f4xx_hal-75c8d3fa54dd8e15.rlib" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/librand_core-a547618338c14625.rlib" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libcast-1346153f242d37ad.rlib" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libstm32f4-d258cb858fbaa62f.rlib" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libembedded_hal-b7cbcb0268d17305.rlib" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libvoid-2dabe3b80173d027.rlib" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libnb-d97f13218ab65d56.rlib" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libnb-22d098c14b2cb118.rlib" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libcortex_m_semihosting-8456019f1bf0dae1.rlib" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libcortex_m_rt-aced26f2ef700965.rlib" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libr0-773be1afbebc52de.rlib" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libcortex_m-9b89eb9831ef9973.rlib" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libvolatile_register-24079378c8e136a9.rlib" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libvcell-fb352589e1f3b95f.rlib" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libbare_metal-2a68be6ea40cd851.rlib" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libaligned-20d500017c4a3b09.rlib" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libas_slice-6fadded5466b728a.rlib" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libstable_deref_trait-a625275795e84c76.rlib" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libgeneric_array-06405a977e0de214.rlib" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libgeneric_array-93fcf103a6a95b71.rlib" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libtypenum-c0a19d0e8b50d874.rlib" "--start-group" "/Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libpanic_halt-a51b264357e80e04.rlib" "/Users/chandamon/.rustup/toolchains/nightly-x86_64-apple-darwin/lib/rustlib/thumbv7em-none-eabihf/lib/librustc_std_workspace_core-998b97e0d6b5f728.rlib" "/Users/chandamon/.rustup/toolchains/nightly-x86_64-apple-darwin/lib/rustlib/thumbv7em-none-eabihf/lib/libcore-369bbf83412ca20c.rlib" "--end-group" "/Users/chandamon/.rustup/toolchains/nightly-x86_64-apple-darwin/lib/rustlib/thumbv7em-none-eabihf/lib/libcompiler_builtins-635853e92b6c58cf.rlib" "-Tlink.x" "-Tdefmt.x" "-Bdynamic"
  = note: rust-lld: error: undefined symbol: _defmt_acquire
          >>> referenced by export.rs:49 (/Users/chandamon/.cargo/git/checkouts/defmt-7f5b74b4e6ff55d4/c97bb56/src/export.rs:49)
          >>>               defmt-36442d255a9c07bc.defmt.111d6lvd-cgu.0.rcgu.o:(defmt::export::acquire::hca9f12e8e06bcf1b) in archive /Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libdefmt-36442d255a9c07bc.rlib

          rust-lld: error: undefined symbol: _defmt_release
          >>> referenced by export.rs:60 (/Users/chandamon/.cargo/git/checkouts/defmt-7f5b74b4e6ff55d4/c97bb56/src/export.rs:60)
          >>>               defmt-36442d255a9c07bc.defmt.111d6lvd-cgu.0.rcgu.o:(defmt::export::release::hac8064c98fc1cd52) in archive /Users/chandamon/Documents/freshup/atom/ok/target/thumbv7em-none-eabihf/debug/deps/libdefmt-36442d255a9c07bc.rlib

error: aborting due to previous error; 5 warnings emitted

error: could not compile `ok`.

To learn more, run the command again with --verbose.

```

jonas-schievink, 2020-09-17:
Can you provide the code needed to reproduce this?

jonas-schievink, 2020-09-17:
Just to make sure, you did define a [`#[global_logger]`](https://defmt.ferrous-systems.com/global-logger.html), right?

chenyanming, 2020-09-17:
Thanks for your help. I am farely new to rust, but I can not find any example of how to add a global_logger. 

I try:
```
use defmt::{Logger, Write, global_logger};
use core::ptr::NonNull;
///
#[global_logger]
struct MyLogger;
///
unsafe impl Logger for MyLogger {
    fn acquire() -> Option<NonNull<dyn Write>> {
// # todo!()
        // ...
    }
    unsafe fn release(writer: NonNull<dyn Write>) {
// # todo!()
        // ...
    }
}
```

Which I found in src/lilb.rs. But it can not work.

chenyanming, 2020-09-17:
I tried to add 
`use defmt_rtt as _;` in `main.rs`. 

`defmt::info!("Hello, world!");` in main function did not work.

But if I used `rtt_target`, it works fine. Do you know why?

chenyanming, 2020-09-17:
My `Cargo.toml`.
```
[package]
authors = ["DAMON"]
edition = "2018"
readme = "README.org"
name = "ok"
version = "0.1.0"

[dependencies]
embedded-hal = "0.2"
nb = "0.1.2"
cortex-m = "0.6.0"
cortex-m-rt = "0.6.10"
cortex-m-semihosting = "0.3.3"
panic-halt = "0.2.0"
heapless = "0.5.5"
# rtt-target = { version = "0.2.2", features = ["cortex-m"] }
# panic-itm = "0.4.0"
# aux5 = { path = "auxiliary" }

# Uncomment for the panic example.
# panic-itm = "0.4.1"

# Uncomment for the allocator example.
# alloc-cortex-m = "0.3.5"

# Uncomment for the device example.
# Update `memory.x`, set target to `thumbv7em-none-eabihf` in `.cargo/config`,
# and then use `cargo build --examples device` to build it.
# [dependencies.stm32f3]
# features = ["stm32f303", "rt"]
# version = "0.7.1"

# this lets you use `cargo fix`!
# [dependencies.stm32f4]
# version = "0.11.0"
# features = ["stm32f407", "rt"]

[features]
# set logging levels here
default = [
  "defmt-default",
  # "dependency-a/defmt-trace",
  "defmt-trace",
]

# do NOT modify these features
defmt-default = []
defmt-trace = []
defmt-debug = []
defmt-info = []
defmt-warn = []
defmt-error = []

[dependencies.defmt]
git = "https://github.com/knurling-rs/defmt"
branch = "main"

[dependencies.defmt-rtt]
git = "https://github.com/knurling-rs/defmt"
branch = "main"

[dependencies.panic-probe]
git = "https://github.com/knurling-rs/probe-run"
branch = "main"

[dependencies.stm32f4xx-hal]
version = "0.8.3"
features = ["rt", "stm32f407"] # replace the model of your microcontroller here

[[bin]]
name = "ok"
test = false
bench = false

[profile.dev]
codegen-units = 1
debug = 2
debug-assertions = true # <-
incremental = false
opt-level = 3 # <-
overflow-checks = true # <-

[profile.release]
codegen-units = 1
debug = 2
debug-assertions = false # <-
incremental = false
lto = 'fat'
opt-level = 3 # <-
overflow-checks = false # <-

# do not optimize proc-macro crates = faster builds from scratch
[profile.dev.build-override]
codegen-units = 8
debug = false
debug-assertions = false
opt-level = 0
overflow-checks = false

[profile.release.build-override]
codegen-units = 8
debug = false
debug-assertions = false
opt-level = 0
overflow-checks = false
```

jonas-schievink, 2020-09-17:
How are you running the firmware? If you use probe-run, you have to pass `--defmt` to it, or it won't decode the received messages.

chenyanming, 2020-09-17:
I am using `cargo run` to run the firmware. Here is my `.cargo/config` file

```
# uncomment this to make `cargo run` execute programs on QEMU
# [target.thumbv7em-none-eabihf]
# runner = "qemu-system-arm -cpu cortex-m4 -machine lm3s6965evb -nographic -semihosting-config enable=on,target=native -kernel"

[target.'cfg(all(target_arch = "arm", target_os = "none"))']
# uncomment ONE of these three option to make `cargo run` start a GDB session
# which option to pick depends on your system
# runner = "arm-none-eabi-gdb -q -x openocd.gdb"
# runner = "gdb-multiarch -q -x openocd.gdb"
# runner = "gdb -q -x openocd.gdb"
runner = "probe-run --chip STM32F407ZGTx"

rustflags = [
  # LLD (shipped with the Rust toolchain) is used as the default linker
  "-C", "link-arg=-Tlink.x",

  # if you run into problems with LLD switch to the GNU linker by commenting out
  # this line
  # "-C", "linker=arm-none-eabi-ld",

  # if you need to link to pre-compiled C libraries provided by a C toolchain
  # use GCC as the linker by commenting out both lines above and then
  # uncommenting the three lines below
  # "-C", "linker=arm-none-eabi-gcc",
  # "-C", "link-arg=-Wl,-Tlink.x",
  # "-C", "link-arg=-nostartfiles",

  # defmt logging
  "-C", "link-arg=-Tdefmt.x",
]

[build]
# Pick ONE of these compilation targets
# target = "thumbv6m-none-eabi"    # Cortex-M0 and Cortex-M0+
# target = "thumbv7m-none-eabi"    # Cortex-M3
# target = "thumbv7em-none-eabi"   # Cortex-M4 and Cortex-M7 (no FPU)
target = "thumbv7em-none-eabihf" # Cortex-M4F and Cortex-M7F (with FPU)
```

I tried to add `--defmt`, but it still failed.
```
warning: 5 warnings emitted

    Finished dev [optimized + debuginfo] target(s) in 0.32s
     Running `probe-run --chip STM32F407ZGTx --defmt target/thumbv7em-none-eabihf/debug/ok`
error: Found argument '--defmt' which wasn't expected, or isn't valid in this context

USAGE:
    probe-run [FLAGS] <ELF> --chip <chip>

For more information try --help
```

chenyanming, 2020-09-17:
Oh, I found I did not run the following to enable `defmt` feature:
```
cargo install \
    --git https://github.com/knurling-rs/probe-run \
    --branch main \
    --features defmt
```

jonas-schievink, 2020-09-17:
Yeah, right now you need to enable the `defmt` feature. This will be simplified once defmt is ready to be released.

chenyanming, 2020-09-17:
Thanks your great help. It works perfectly now.

closed 2020-09-17 by chenyanming