knurling-rs/defmt #173: Use of bincode for the serialization

almusil, 2020-09-21:
Maybe I have missed something but would it be viable option to use https://github.com/bincode-org/bincode-core for the serialization? It can be extended with leb128. I see huge benefit in using serde. 
WDYT?

japaric, 2020-09-21:
We rely on symbol names for string compression; this requires creating a `#[link_name = ".."]` attribute whose argument is either the full struct (`"struct S {{ field: {:u8} }}"`, better compression) or at least a field name. We cannot construct neither with the `serde` framework because that will pass us a `&str` value but constructing the `link_name` attribute requires a **string literal**. (There are more details about this in the [design & impl notes](https://defmt.ferrous-systems.com/design.html) section of the book)

almusil, 2020-09-21:
Ah that makes sense, thanks for the explanation.

closed 2020-09-21 by almusil