knurling-rs/defmt #179: Aggressive inlining is a problem

spcan, 2020-09-22:
Selecting a debug string using a `match` statement produces a very aggressive inlining in the code.
Example:
```
match (internal_condition) {
    0 => info!("The condition was not met"),
    1 => info!("The condition was half met"),
    2 => info!("The condition was full met"),
    // ... There are 9 conditions in this check
    _ => (),
}
```
Currently this condition matching embeds a lot into this function. The total size of this function ends up being 11.2 kB.

When, instead of using direct logging, I tried to use a `struct` with a custom derived `Format` , the size of the function gets reduced to around 3.9 kB (still big, but at least more reasonable). It still embeds the whole routine into the function with every `info!` call but now it's not once on each match arm.

Example:
```
#[repr(C)]
pub struct LogHolder(u8);

impl Format for LogHolder {
    fn format(&self, f: &mut defmt::Formatter) {
        match self.0 {
            0 => defmt::export::write!(f, "The condition was not met"),
            // ....
        }
    }
}
```

Is this inlining behaviour by design? If this is not, maybe there is a way to force the compiler to not inline all the routine into the function, but keep it separated, even if it may hinder performance a bit. Currently it is very likely that it will bloat up and fill the whole FLASH space with a small amount of debug calls.

If you need more info, please tell me, I'm happy to provide binaries and disassembly if needed.

PD: Just to clarify, in the function I use there are two matches, one with 9 arms and the other with 7. Currently the code is inlined and repeated all 16 times (plus another single debug call). Reducing it to 3 calls using the structs makes it still inline the routine 3 times.

japaric, 2020-11-09:
I think we can improve code reuse in the case of logging a single string literal by adding a function call.
Given this code:
``` rust
info!("hello");
info!("world");
```
Right now it expands into:
``` rust
// "hello"
if true && true || !true && true {
    match () {
        () => {
            let ts = defmt::export::timestamp();
            if let Some(mut _fmt_) = defmt::export::acquire() {
                _fmt_.istr(&defmt::export::istr({
                    #[link_section = "<omitted>"]
                    #[export_name = "<omitted>"]
                    static DEFMT_LOG_STATEMENT: u8 = 0;
                    &DEFMT_LOG_STATEMENT as *const u8 as usize
                }));
                _fmt_.leb64(ts);
                _fmt_.finalize();
                defmt::export::release(_fmt_)
            }
        }
    }
}
// then repeat for "world"
```
The idea would be to expand it to something like this:
``` rust
// "hello"
if true && true || !true && true {
    defmt::export::log_no_args({
        #[link_section = "<omitted>"]
        #[export_name = "<omitted>"]
        static DEFMT_LOG_STATEMENT: u8 = 0;
        &DEFMT_LOG_STATEMENT
    });
}
// "world"
if true && true || !true && true {
    defmt::export::log_no_args({
        #[link_section = "<omitted>"]
        #[export_name = "<omitted>"]
        static DEFMT_LOG_STATEMENT: u8 = 0;
        &DEFMT_LOG_STATEMENT
    });
}
```

Then the compiler (or rather, LLVM) would at least have the opportunity to *not* inline the `log_no_args` function. Though whether inlining happens or not depends on the opt-level setting, the LLVM version and who knows what else.

I'm not sure how much this approach can be generalized. It doesn't quite scale for log statements that take primitive arguments (`{:u8}`)  and could even be counter-productive for `{:?}` arguments if we try to use generic functions there (e.g. `fn log_2_args(impl Format, impl Format)` may be monomorphized as many times as the log macros are used)

jonas-schievink, 2020-11-09:
Another small thing to try is to put the "get timestamp, acquire logger, X, finalize, release logger" pattern into its own function that takes a `&dyn FnMut` representing the `X` part.

japaric, 2020-11-09:
I'm not sure how good LLVM is at deduplicating / unifying closures. Using `&mut dyn FnMut` may produce as many trait objects as log statements.

Urhengulas, 2022-02-01:
Was this addressed by https://github.com/knurling-rs/defmt/issues/492?

spcan, 2022-02-03:
The defmt crate has changed too much for this issue to be relevant, IMO this issue can be closed

closed 2022-02-04 by Urhengulas