knurling-rs/defmt #180: Linking with GCC GCs DEFMT_LOG_STATEMENT symbols

mattico, 2020-10-02:
rustc 1.48.0-nightly (ef663a8a4 2020-09-30)
gcc version 9.3.1 20200408 (release) (GNU Arm Embedded Toolchain 9-2020-q2-update)

When running with `probe-run --defmt` I always see `  (HOST) WARN  insufficient DWARF info; compile your program with `debug = 2` to enable location info` and no messages are printed. However I am configured to produce debuginfo and I can see the debuginfo and defmt sections in the binary.

My `.cargo/Config.toml`:

```toml
[target.thumbv7em-none-eabihf]
runner = "probe-run --chip stm32h743bi --defmt"

[build]
target = "thumbv7em-none-eabihf" # Cortex-M4F and Cortex-M7F (with FPU)
rustflags = [
  "-C", "linker=arm-none-eabi-gcc",
  "-C", "link-arg=-Wl,-Tlink.x",
  "-C", "link-arg=-nostartfiles",
  "-C", "link-arg=-specs=nano.specs",
  "-C", "link-arg=-lc",
  "-C", "link-arg=-mcpu=cortex-m7",
  "-C", "link-arg=-mfpu=fpv5-d16",
  "-C", "link-arg=-mfloat-abi=hard",
  "-C", "link-arg=-mthumb",
  "-C", "link-arg=-Wl,-Tdefmt.x",
  "-C", "link-arg=-g",
  "-C", "inline-threshold=225",
]

[unstable]
features = ["host_dep"]
```

My `Cargo.toml`:

```toml
[features]
default = ["defmt-default"]
defmt-default = []
defmt-trace = []
defmt-debug = []
defmt-info = []
defmt-warn = []
defmt-error = []

[profile.release]
codegen-units = 1 # better optimizations
opt-level = 3
debug = 2 # symbols are nice and they don't increase the size on Flash
lto = true # better optimizations

[profile.dev.package."*"]
opt-level = "s"
```

Adding logging to defmt_elf2table I can see that all of the `DEFMT_LOG_STATEMENT` symbols have been GCd.

Any suggestions?

nosense-cz, 2020-10-20:
The same problem occurs when linking by arm-none-eabi-ld

jonas-schievink, 2020-10-20:
This probably means that LLD will at some point try to match behavior of GNU LD and also break

japaric, 2020-10-23:
linking a minimal application with GNU LD (or GCC) was not working for me but it's working with PRs #215 and PRs #216 . GNU LD / GCC may want to try again after those PRs have been merged.

Lotterleben, 2020-10-30:
hi @mattico , @nosense-cz  could you check if it works for you now?

jonas-schievink, 2020-11-02:
@nosense-cz You need to use 28c1afb on the firmware, `cargo update -p defmt --precise 28c1afb` might to the job.

nosense-cz, 2020-11-02:
I rebuild probe-run with updated cargo.toml to point to current defmt "main" branch. With that I can successfully see the log messages when linking by arm-none-eabi-ld and also by by arm-none-eabi-gcc. I consider this fixed.

jonas-schievink, 2020-11-02:
Great! Closing then.

closed 2020-11-02 by jonas-schievink