knurling-rs/defmt #182: Expose custom message tags to users

jonas-schievink, 2020-10-07:
https://github.com/knurling-rs/defmt/pull/175 adds a tag to every message. We use it to store the log level, but downstream libraries could use it to mark messages for consumption by other host-side tooling. We should come up with an API to set a custom tag.

japaric, 2021-06-23:
In my opinion this (adding custom tags to messages) should wait until we have a *stable* API to decode defmt messages or at the very least JSON output (knurling-rs/probe-run#178); otherwise there's no stable way to use the messages from tooling or similar. Would you agree @jonas-schievink ?

Unless this meant tagging messages for use by knurling-rs tooling that we maintain.

adityashah1212, 2023-08-03:
I like this idea. It would be great if we could convert this into a structured logging use case, where instead of having `format string` + `data`, we would have `static string message` + `tag=value list`. This can then be used on host side for multitude of things like, multiplexing streams, filtering data as needed, analyzing logs for detecting potential failure and so on.

It can also be used to include build numbers as part of the log message to ensure logs are being parsed by using the right binary. This can happen in cases where logs are locally stored and fetched as needed and logs could be a combination between previous version of the firmware and a newer version that was loaded as part of OTA.

adityashah1212, 2025-06-29:
I thought about this and there is a way to implement this without breaking existing API. The way we could achieve this by allowing less number of `{}` in formatting string compared to passed arguments. This way any fields that don't match a corresponding set of braces in formatting string becomes a `tag=value` pair in the formatting JSON symbol. This way the parser can easily handle these fields. An example would be as follows
```rust
let value = 10u8;
// This should generate the regular format string expecting 1 value and two tags `tag` and `value`
write!("I have this value {value=u8}", value, tag=value, value);
```

The above should roughly change to the following
```rust
#[export_name = "{ \"package\": \"test-package\", \"data\": \"I have this value {value=u8}\", \"disambiguator\": \"1379186119\", \"attributes\": [ \"tag\", \"value\" ] }"]
#[link_section = ".."]
static SYM: u8 = 0;
```
I am unsure of how defmt figures out the size of each field that is being added to the formatting string if type hint isn't provided as I haven't yet delved that deep in the implementation, but I believe the idea should be obvious. If more metadata is required, `attributes` can be an array of JSON objects instead of array of strings.

This should similarly translate to a JSON to contain an optional field of `attributes` containing JSON object with two keys of `tag` and `value`, like so
```JSON
{ "data": "I have this value 24", "host_timestamp": 123456,"location": { /*snip*/ }, "attributes": [ { "tag": "tag", "value": 24 }, { "tag": "value", "value": 24 } ] }
```
This is not breaking for on-wire protocol as we will still be serializing only data in the same sequence as they appear.

It won't break API as the extra fields are optional and existing ones that comply with strict requirements of number of `{}` should match the number of arguments will still work.

As long as we minor version to `1.1`, we can ensure that the decoder and `defmt` is bumped at the same time to ensure that older decoder is not used to decode newer messages. As for JSON parser for log messages, same thing can be applied by updating the schema to version `1.1`

jonathanpallant, 2025-07-01:
So in this example,

> tag=value

is interpreted as a JSON dictionary entry of `"tag": "value"`? That seems like it might cause confusion, given that a plain `value` prints the *value* of the expression `value`, rather than its literal name.

I think if people really want a mechanism for putting arbitrary JSON key/value pairs into the symbol name, we'd need to add a new set of macros to do it.