knurling-rs/defmt #185: Functions in defmt::Write to allow encoding and framing

MathiasKoch, 2020-10-08:
As mentioned in #168, it would be awesome to have some functions along the lines of `start_of_frame` and `end_or_frame`, to allow implementing stuff like CRC and framing (eg. COBS).

I am opening this issue, as I think this is hidden and forgotten in the other issue, as it came up as a side-note.

Preferably the `start_of_frame` would contain information about the total length of the full frame to come.

jonas-schievink, 2021-01-13:
@japaric hmm, why is this a breaking change? Couldn't we provide an empty default implementation in the trait?

japaric, 2021-01-14:
> hmm, why is this a breaking change? 

Hmm, I forget the reason. I was probably thinking that adding a frame to the defmt stream is a breaking change and it is at the level of the "transport" crate (e.g. defmt-rtt) but it's not a breaking change in the `defmt` crate.

> Couldn't we provide an empty default implementation in the trait?

yes, that change in `defmt` should be non-breaking .

mattico, 2021-04-22:
What's the motivation behind putting the framing into the `Write` trait rather than a level up? Whatever calls `write` can just do the COBS encoding itself in a streaming fashion before it calls `write`. Then the encoding can be shared amongst the transports and doesn't need to be re-implemented.

Do some transports already have their own framing protocol and thus don't need to have one implemented on top?

jonas-schievink, 2021-04-22:
Framing isn't strictly necessary to use defmt, so we wanted to make it optional. The framing algorithm to use also might be something the user wants to control (for example, COBS vs. SLIP is a tradeoff between worst-case overhead and RAM usage).

mattico, 2021-04-22:
Right, the transport implementation is pluggable so it's a more flexible place for modifications.

closed 2023-06-22 by MathiasKoch