knurling-rs/defmt #205: feature request: treat {} the same as {:?}

Dirbaio, 2020-10-21:
Defmt doesn't support `{}`, it requires you to write `{:?}`. This is the #1 annoyance when porting code from `std::fmt` to `defmt`. It would be great if it treated both the same.

If a codebase wants to keep compatibility with both `defmt` and `std::fmt` it's even worse: a find+replace is not enough, since `{}` and `{:?}` mean different things in `std::fmt` (one uses `Display`, the other uses `Debug`), so this would change the behavior of the `std` builds.

japaric, 2020-10-22:
I'm a bit unsure about this feature.

On one hand, people should be using e.g. `{:f32}` instead of `{:?}` to save bandwidth; in practice I think most people don't, because coming from `core::fmt` they'll try either `{}` or `{:?}` and since the latter works they will use that everywhere. Making `{}` work doesn't help with that (doesn't lead people towards using `{:$type}`); it seems to me that making `Format` (`{:?}`) more ergonomic (`{}` requires less typing) will actually have the opposite effect (people will always use the `Format` indirection).

On the other hand, accepting `{}` work does make it easier to switch between say `log` and `defmt` (example below) but it is not strictly required for that kind of inter-op. A third-party proc-macro could accept the `core::fmt` syntax and translate the `{}`s and `{:?}`s to `defmt`'s `{:?}`s based on some Cargo feature. That proc-macro idea has been mentioned before in this issue tracker. (Though I'm unsure whether it *always* makes sense to use `Format` on non-primitives types that were intended to be formatted with `fmt::Display` since `Format` is closer to `fmt::Debug` but that may be a different discussion)

``` rust
#[cfg(feature = "use-core-format")]
use log::info;

#[cfg(not(feature = "use-core-format"))]
use defmt::info;

info!("x={}", 42);
```

Ideally, I would like to have `defmt`'s `{}` do the zero overhead (no `Format` indirection) encoding of primitive types and get rid of `{:u8}` and others but I don't think this can be implemented because we need to put the type info in the string that goes in the side table (ELF symbol table) and proc-macros cannot access type information, which is why no-indirection encoding requires the user to provide the type to the macro. (wait, I just thought of a way to implement that; it involves getting the argument type from the DWARF info; unsure if it will actually work though)

jonas-schievink, 2020-11-13:
I noticed that formatting a `&str` with `{:str}` or `{:?}` currently uses the unescaped formatting, like `{}` with `core::fmt`. This isn't always appropriate, and sometimes you *do* want the escaped `{:?}` format from `core::fmt`, so maybe we should find a way to make that possible? It wouldn't require any wire format or API changes, just host changes.

japaric, 2021-01-05:
Done as part of RFC #270

closed 2021-01-05 by japaric