knurling-rs/defmt #208: corrupted stream when mixing bools, options and arrays

Dirbaio, 2020-10-22:
```rust
    #[derive(defmt::Format)]
    pub struct DhcpRepr {
        pub broadcast: bool,
        pub a2: Option<[u8;6]>,
        pub a3: Option<[u8;6]>,
    }

    let mut dhcp_repr = DhcpRepr {
        broadcast: true,
        a2: Some([1, 2, 3, 4, 5, 6]),
        a3: None,
    };
```

if you stick it in the qemu firmware, it outputs this:

```
Error: couldn't decode all data (remaining: [70, 49, 6f, d, 1, b, 1, 2, 3, 4, 5, 6, d, 0, 1])
```

reproduces on latest `main` branch

jonas-schievink, 2020-11-04:
I'm seeing this instead:

    0.000087 INFO print of death: DhcpRepr { broadcast: false, a2: Some([1, 3, 4, 5, 6, 15]), a3: 1 }

(which is arguably worse :smile: )

jonas-schievink, 2020-11-04:
The derived `Format` impl contains a `_fmt_.finalize();` call, which is wrong. `finalize` should only be called by the logging macros once the whole frame has been encoded, otherwise we end up with additional compressed bool fields in the message.

jonas-schievink, 2020-11-04:
Hmm, that does not seem to be the cause of this issue though. The same bug is in the decoder, so things end up working out fine.

jonas-schievink, 2020-11-04:
Ah, the `Format` impl of `bool` also had a bug. Opened https://github.com/knurling-rs/defmt/pull/228 to fix this.

commit 38b7b6e7 (2020-11-04): Add encoding test for #208

Closes #208
closed 2020-11-04 by jonas-schievink