knurling-rs/defmt #210: corrupted stream with custom Format impl using `|`

Dirbaio, 2020-10-22:
```rust

struct Foo {
    a: u32,
    b: u32
}

impl Format for Foo {
    fn format(&self, fmt: &mut defmt::Formatter) {
        defmt::export::write!(fmt, "{:u32} | {:u32}", self.a, self.b);
    }
}

defmt::info!("asdfasdf: {:?}", Foo{ a: 1, b: 2});
```

(is doing custom Format impls like that  actually supported?)

japaric, 2020-10-23:
> is doing custom Format impls like that actually supported?

this is not supported at the moment. we want to expose a syntax-checked (to prevent what you are seeing there) `write!` macro to end users at some point but that needs some design work. `export::write!` is a "private", unstable implementation detail that shouldn't be used. ("private" as in: it's not in the API docs; it has to be public due to how proc macros work though)

Dirbaio, 2020-10-23:
Okay! Thanks for confirming, that's what I feared :) 

It would be cool to support it. I'm working on adding defmt support to all the smoltcp types, and this is super handy with all the basic types like IpAddress and EthernetAddress.

Maybe the new JSON metadata could be used for this? Interned strings could be annotated with "This is an enum" or "This is a custom write! call" so the decoder parses them differently.

jonas-schievink, 2020-11-04:
> Maybe the new JSON metadata could be used for this? Interned strings could be annotated with "This is an enum" or "This is a custom write! call" so the decoder parses them differently.

Yeah I think this might be a good way forward – it allows users to write completely custom `Format` impls, but still lets us detect `#[derive]`d impls and do structured logging for those.

jonas-schievink, 2020-11-10:
The `write!` macro is being discussed in https://github.com/knurling-rs/defmt/issues/104, so closing in favor of that.

closed 2020-11-10 by jonas-schievink