knurling-rs/defmt #223: Allow using on x86_64

Dirbaio, 2020-11-01:
The library currently hardcodes test-only behaviors for x86_64. This makes it impossible to actually use it on x86_64. 

(The opposite is also true, I think: tests would break if you run them on a Raspberry Pi for example)

jonas-schievink, 2020-11-10:
Seems reasonable to make the tests use some other mechanism to work. Not quite sure how that would look, but it's all private APIs, so there's no risk of breaking things.

Dirbaio, 2020-11-10:
I think the standard way to do it is with `#[cfg(test)]`. 

Is there any reason it's not done that way? If not I might take a look at doing it.

jonas-schievink, 2020-11-10:
`#[cfg(test)]` does not work for API that is used by integration tests, I think that's why we don't use it. Unless it was an oversight.

japaric, 2020-11-11:
If `cfg(test)` doesn't work there's the option to add an internal, e.g. "test", Cargo feature and enable it when running tests. ('internal' as in not meant to be enabled by third party library / applications)

PR #291 (2020-12-01): Enable test-only code via feature instead of target_arch.
Fixes #223 

- Replace `target_arch = "x86_64"` with `feature = "unstable-test"` everywhere
- Add features to `Cargo.toml`s
- Add `--features unstable-test` in CI script
closed 2020-12-01 by japaric