knurling-rs/defmt #249: Performance - raw cycle count of an info! call

bradleyharden, 2020-11-13:
Hi,

I've been trying to find a solution that will let me log within a task running at 30 kHz. That gives me 3200 cycles at 96 MHz. At first, I was using `core::fmt` with `rtt_target`, but formatting was taking 2500+ cycles. I switched to `ufmt`, which brought that down to 650-700 cycles. Finally, I tried `defmt`, thinking it would reduce the cost further.

Making the switch was a little complicated, because `probe-rs` doesn't yet have full support for the SAMD series chips. I've had to piece together some branches from @jacobrosenthal to get it all working. It works, but it's slow. The speed maxes out at `--speed 100` on my J-Link, and I think we can bump it to `--speed 200` on a CMSIS-DAP probe.

Switching to `defmt` has been great overall, but performance didn't improve. I'm using the following code for a performance test:

```rust
let now = DWT::get_cycle_count();
info!("now = {:u32}", now);
let diff = DWT::get_cycle_count() - now;
info!("diff = {:u32}", diff);
```

I'm getting `diff` values of around 800. I discussed this with @Dirbaio in the `probe-rs` Matrix chat. He took at look at the asm output of an `info!` call and wasn't surprised that it takes 800 cycles. I suppose that means there's not much hope of improving performance in the near future, but I wanted to bring up the issue for discussion.

For my case, logging a single integer takes 25% of my budget, which is just not practical. I guess logging isn't really going to be possible for me. I can always slow things down, but it would be ideal if I could do it all in real-time.

Do you have any thoughts or suggestions?

japaric, 2020-11-13:
Some ideas to improve the perf of a *single* `info!` invocation (none of these are currently supported / implemented)

`defmt-rtt` uses the RTT protocol which is a circular buffer with read / write pointers. The read pointer is advanced from the host on reads; the write pointer is advanced by the firmware on writes. The firmware checks the read pointer and compares it to the write pointer to avoid overwriting unread data. Instead of the RTT protocol I have used before a circular buffer with only a write pointer to send defmt data; I recall 3-5 microseconds per `info!` call on a 64 MHz CPU; downside of this approach is that unread data can be overwritten and this corrupts the stream of defmt log frames

omit the timestamp from the `info!` call: this avoid accessing a global / singleton *and* LEB encoding the timestamp

`info!` implicitly *acquires* and *releases* the global logger. If there was an API to acquire it you could reduce the cost of consecutive `info!` invocations:

``` rust
let logger = defmt::acquire();
info!(logger, "x={:u32}", x);
info!(logger, "y={:u32}", y);
info!(logger, "z={:u32}", z);
drop(logger); // release
```

---

Depending on your application, you may be able to buffer those `u32` values into a heapless Vec and log them in bulk after a few (e.g. 10 or 100) iterations.

---

Even if you improve the device side perf of `info!` the bottleneck is usually not there but the link between the probe and the host. The link is usually USB Full Speed (FS) HID 2.0 which is limited to 64B packets and 1 millisecond delay per packet. Theoretical throughput is 64 KB/s but you need to consider the overhead of e.g. CMSIS DAP (which uses request-response so 2 HID packets per probe command) and the overhead of the RTT protocol so in practice the number is much lower. Logging a `u32` at 30kHz is at the very *least* 120KB/s of throughput which is above what USB FS HID can handle.

I think that to log at 30kHz you'll need a USB High Speed (HS) probe or a probe with built-in memory to buffer log messages.

I guess that if you have enough memory you could defmt log everything to RAM and inspect the log after your application finishes (*if* your application finishes; also this is a unimplemented feature)

bradleyharden, 2020-11-13:
Good point about overall bandwidth. I forgot about that. I can get 240 kB/s with RTT and my J-Link probe, but that uses the proprietary J-Link code, and it requires the J-Link driver.

Really, that 30 kHz process was hanging, and I was trying to find out why. I was hoping to log until it hangs, which was only a few calls in. I wasn't thinking about logging at that rate for long.

Still, at 800 cycles, even occasional logging could break my budget. I like some of the ideas you proposed. I'll need to see what my other use cases are before I decide which direction to go.

Dirbaio, 2020-11-13:
This is somewhat related to #179 (bigger code means flash bloat and execution time bloat).

One possible solution to improve both would be:
- Construct log frame in a fixed-size stack buffer
- Do a single function call to write it out. That function will do acquire+write+release, and it probably can be optimized now that everything's a single write.

I'm pretty sure this would be much faster
- Only 1 RTT write call instead of multiple (currently there's one for timestamp, one for interned string, and one (or more) for each argument)
- Compiler knows the buffer offsets at compile time, so it can probably generate much more compact code.

Downside is that you need to know the buffer size and handle overflow...

japaric, 2020-11-13:
> One possible solution to improve both would be:

I tried this and it didn't end up being faster than the "read-pointer-less RTT" because of ...

> Downside is that you need to know the buffer size and handle overflow...

overflow handling to not lose data and probably due to the "double memcpy": you write data to a stack buffer first (that's like one memcpy) and then memcpy that into the RTT buffer. But it would good to check the perf numbers again though

Dirbaio, 2020-11-13:
:(

Idea num 2: Maybe not using LEB128 would help? That way most write offsets to build the log frame are compile-time known (even all for simple logs).

Idea num 3: use a zero-removing compression if removing LEB128 makes frames too big. https://github.com/knurling-rs/defmt/issues/199#issuecomment-713505349 . This could be done "in place of" the second memcpy you mention (from stack buffer to rtt buffer), so it wouldn't add that much extra cost. Also centralizing the compression in a single place makes it easier to make it optional if you really need the extra speed.

Idea num 4: Maybe it's fine to drop log frames if they're bigger than the buffer? A 256-byte buffer would be enough for 99.9% of log statements. If it doesn't fit, don't try to handle it, just log some error like "Frame dropped due to too big" instead. User could then change it to a `info_buf!(1024, "foo {:?}", my_huge_array)`.

bradleyharden, 2020-11-28:
FWIW I like the idea of a fixed buffer size that can be manually increased if necessary. To me, it seems reasonable to heavily optimize the primary use cases, as long as there is an escape hatch.

Urhengulas, 2022-02-01:
Was this addressed by https://github.com/knurling-rs/defmt/issues/492?

bradleyharden, 2022-02-01:
I have no idea. I haven't really kept up with the defmt ecosystem. But either way, I don't know that this issue needs to remain open. The discussion ran its course. Feel free to close.

Urhengulas, 2022-02-01:
Kk. Thank you for your answer!

closed 2022-02-01 by Urhengulas