knurling-rs/defmt #251: `global_logger` + printer for non-realtime logging?

twitchyliquid64, 2020-11-13:
Afaict all the existing global loggers assume you are one with the uC via RTT or semihosting or similar. Are you aware of any logger, or plans for one, that would support yeeting the log frames into memory / SD card, and then decoding them later?

twitchyliquid64, 2020-11-13:
One potential challenge: with `probe run` or similar its quite impossible to decode the log frames against the wrong ELF, but thats definitely an easy mistake to make when copying around files. Is there any kind of 'unique marker' or similar that gets written to the log frames at the beginning, and is also present in the ELF?

MathiasKoch, 2020-11-15:
I have created this to eventually be used at my workplace: https://github.com/BlackbirdHQ/defmt-persist 

I would be happy to see it gain some OS adoption, and are more than happy to take contributions.

It is still a WIP, and currently hangs a bit on a few blockers, main one being:
- embedded-storage repo (in progress: https://github.com/rust-embedded/embedded-hal/pull/241#issuecomment-726601960)
- Framing options in defmt (https://github.com/knurling-rs/defmt/issues/185)

Basically I make use of an embedded-storage trait to work as storage space (stack/heap, SD-card, file system, flash storage etc), where i implement a BIP style circular buffer on top, with the first N bytes allocated as header.
I then store each log message with a CRC checksum in a COBS encoded format, making it easy to distinguish frames from each other, and some functions to retreive the log frames afterwards (how you then transfer the frames are up to you. Could be RTT, semihosting, network, etc. In my case I am using MQTT)

I have a small static queue to make sure we can move the long running workload of writing to storage, into some low priority idle function, inspired a bit by cortex-m-funnel.

So basically:
```
#[global_logger] ---> queue ---> Storage ---> Transport
                              ^ In idle     ^ On request
```

twitchyliquid64, 2020-11-22:
Nice! I've started on a global logger that just puts everything into a circular buffer, with the aim that user-code will service the buffer and send it to the computer via UART. It didnt work however and I need to debug it, so I'm keen to use the queue/buffer bits of `defmt-persist` instead :)

My current focus is on the formatting hints PR, but let me know if theres anything I can help with I can circle back to and help out!

MathiasKoch, 2020-11-23:
Yeah, i already added that to `defmt-persist`, but i probably should add a few examples... 

If you have a look in https://github.com/BlackbirdHQ/defmt-persist/blob/master/src/stack.rs, i have basically implemented the `Storage` traits needed for `defmt-persist` on any regular `&'static [u8]`

I do hope we can push for the storage traits to reach some release, where i don't have to tie the crate to a git dependency

jonas-schievink, 2021-01-15:
We have now published `defmt-print`, a command-line tool that decodes defmt frames from stdin. This should be helpful for custom defmt pipelines, including decoding log frames "post-mortem".

jonas-schievink, 2021-01-19:
Given that all pieces you need to implement arbitrary defmt storage and transport solutions, closing this as fixed.

closed 2021-01-19 by jonas-schievink