knurling-rs/defmt #255: Alternatives to Cargo feature proliferation

jonas-schievink, 2020-11-14:
defmt requires that every library using it declare 6 Cargo features to allow choosing logging levels, `defmt-default`, `defmt-trace`, `defmt-debug`, `defmt-info`, `defmt-warn`, `defmt-error`.

It would be nice if there was another solution that didn't require this whole stampede of Cargo features. It might be possible to use a compile-time environment variable for this that works similar to the well-known `RUST_LOG`.

I'm not entirely sure how this would be accomplished though. It seems that we'd need build scripts that inspect the variable and use `rerun-if-env-changed`, and output the right `rustc-cfg` directives. I'm hoping that this can be accomplished without forcing every user to define a `build.rs`.

Dirbaio, 2020-11-14:
If defmt's `build.rs` has `rerun-if-env-changed`, changing the envvar will already cause all defmt-dependent crates to recompile. Not sure if it's OK or too fragile to rely on that though...

jonas-schievink, 2020-11-14:
> If defmt's `build.rs` has `rerun-if-env-changed`, changing the envvar will already cause all defmt-dependent crates to recompile. Not sure if it's OK or too fragile to rely on that though...

Yeah that's true. I don't think you can set `cfg` flags that apply to downstream crates through this though.

Dirbaio, 2020-11-14:
The proc macros can read from env directly

jonas-schievink, 2020-11-14:
That is true, and if combined with `rerun-if-env-changed` there should be no issues with the macros not being pure (since everything that could be affected gets recompiled).

I thought this was going to be more complicated, maybe we are missing something?

japaric, 2020-11-16:
This idea with `rerun-if-env-changed` was brought up in reddit several weeks ago but it has no been explored. My knee reaction when I heard the idea was:
- how does the e.g. `info!` retrieve the crate name of the *caller* site in proc-macro context? Assuming we follow `RUST_LOG` syntax we want `RUST_LOG=panic_probe=error` to be picked up by log calls from the `panic_probe` crate.

- this sounds like it's going to cause more crates to recompile than using Cargo features. Changing the Cargo features of a single crate causes that crate and its reverse-dependencies to be recompiled (that's today mechanism); changing the env var is going to recompile all crates that depend on `defmt` *and* all their reverse-dependencies.

Dirbaio, 2020-11-16:
> how does the e.g. info! retrieve the crate name of the caller site in proc-macro context?

the macro could generate something like this [(playground)](https://play.rust-lang.org/?version=stable&mode=debug&edition=2018&gist=87f0d95297ee93646d171700b22bbfba). I *think* it should get completely evaluated at compile-time. Hopefully it doesn't make the compiler too slow. you can't compare slices in const it seems :(

```rust
    const PATH: &[u8] = core::module_path!().as_bytes();
    const LOG: bool = 
        PATH[0] == b'p' &&
        PATH[1] == b'l' &&
        PATH[2] == b'a' &&
        PATH[3] == b'y' &&
        PATH[4] == b'g' &&
        PATH[5] == b'r' &&
        PATH[6] == b'o' &&
        PATH[7] == b'u' &&
        PATH[8] == b'n' &&
        PATH[9] == b'd';
    
    if LOG {
        println!("Must log!");
    }
```

> this sounds like it's going to cause more crates to recompile than using Cargo features

Aww that's true. Maybe we can do one envvar per crate, like `DEFMT_MYCRATE`, so `mycrate` gets recompiled only when that one changes, not when other `DEFMT_*` changes. This'll require every crate to add a build.rs to use `rerun-if-env-changed`.

Dirbaio, 2020-11-16:
it's a bit nicer with slice patterns. Makes it easier to do "Log everything in `playground::` except `playground::foo`". [link](https://play.rust-lang.org/?version=stable&mode=debug&edition=2018&gist=bc35829634ebae5e78865d0db6216a5c)

jonas-schievink, 2020-11-16:
> how does the e.g. `info!` retrieve the crate name of the _caller_ site in proc-macro context?

The downstream crate name should be accessible via `env::var("CARGO_CRATE_NAME")` in the macro. At least that works with `CARGO_PKG_VERSION`, which I've used in [`pkg-version`](https://docs.rs/pkg-version/).

> this sounds like it's going to cause more crates to recompile than using Cargo features

Yup. Personally I'm willing to accept that tradeoff. If the environment variable change does not affect a crate, theoretically what you get is a 100% reuse incremental build, which shouldn't be too bad (does Cargo enable incremental compilation for crates.io dependencies?).

Dirbaio, 2020-11-16:
> The downstream crate name should be accessible via env::var("CARGO_CRATE_NAME") in the macro.

Ooh, neat! I'd prefer to be able to match on module too, not just on crate. 

We could use the env to get filters for current crate, then generate no code, unconditional code, or the `module_path!()` hack depending on found filters

japaric, 2021-06-24:
I've submitted a PoC implementation in PR #519

closed 2021-10-04 by bors[bot]