knurling-rs/defmt #256: Using defmt on WSL provides a cryptic error message if git is missing

tl8roy, 2020-11-14:
While `libudev` and `libusb` are mentioned as dependencies, `git` is not. As such when trying to install `probe-run` or use defmt in a project you can get:
```
error: failed to run custom build command for defmt-decoder v0.1.1
Error: Os { code: 2, kind: NotFound, message: "No such file or directory" }
```
This was hard to work out until `git` was mentioned in chat.
Once git was installed, it worked fine.

japaric, 2020-11-16:
Thanks for the report. To double check: this was with a `--git` install and not a crates.io install?

In either case, installation should not error if `git` is not installed. `git` is not a dependency of `defmt` or `probe-run`.

tl8roy, 2020-11-16:
This was a crates.io install.
'cargo install probe-run'

harrysarson, 2020-11-16:
It looks like the build script of the defmt-decoder crate shells out to git here:

https://github.com/knurling-rs/defmt/blob/main/decoder/build.rs#L7

japaric, 2020-11-16:
Yeah, the 2 build scripts in this repo should invoke git like [probe-run's build script](https://github.com/knurling-rs/probe-run/blob/519a03a9d70de6c4d1835a02581a581a319eaa59/build.rs#L15) does to avoid error-ing out on git not being installed.

japaric, 2020-11-16:
This may be a more complicated than I hoped.
If we are doing a crates.io install then we should not depend on git at all. That's easy to fix.
If we are doing a git install then we do require the commit hash of the repo checkout by Cargo *even if there's no git command line tool installed on the system*. That's the tricky part. Making the build script depend on the libgit2 crate will be headache to install for Windows users.

japaric, 2020-11-16:
wait, I think we can read the `.git/ORIG_HEAD` file instead of shelling out to git

PR #259 (2020-11-16): make crates.io version not depend on the `git` tool
also error if git version is used but `git` is not installed
prompt the user to install `git` in that case

fixes #256

---

we need `git` when the git version is used because we use the commit hash to check if probe-run is compatible with the firmware-side defmt
instead of panicking when git is not installed we could:
- depend on the libgit2 crate and use that to retrieve the commit hash (from what I remember libgit2 is as, or more, annoying to install as git on Windows (not WSL))
- implement `git rev-parse HEAD` using just the std library.  [... cut]
commit e5cc0de0 (2020-11-16): make crates.io version not depend on the `git` tool

also error if git version is used but `git` is not installed
prompt the user to install `git` in that case

fixes #256
closed 2020-11-16 by japaric