knurling-rs/defmt #260: Rust-Analyzer's "Run test" buttons don't work with defmt-test

japaric, 2020-11-16:
The `#[test]` attribute is picked up by Rust-Analyzer so RA shows "Run Test" buttons but you get an error if you try to use.
Those button will invoke `cargo test --package testsuite --test test -- tests::assert_true --exact --nocapture` which then invokes passes the `tests::assert_true --exact --nocapture` arguments to `probe-run`.
`probe-run` will fail to run as it doesn't recognize those command line arguments.
Interestingly the `mod tests` module is also picked up by RA so it also gets a "Run Test" button; that one also fails to run.

[image]

(migrated from https://github.com/knurling-rs/defmt-test/issues/1 )

Lotterleben, 2020-12-02:
on what is this issue blocked?

japaric, 2020-12-07:
> on what is this issue blocked?

being able to customize the behavior of "Run Test" in Rust Analyzer maybe? RA assumes the standard (`std`) test runner will be used: that's why it passes those extra command line flags (which probe-run doesn't recognize). That's a fair thing to assume given that `#[test]` is only available in `std` on the stable channel. `defmt-test` is *pretending* to be the standard `#[test]` attribute; hence the mismatch. If RA had an option to stop passing these flags then we could fix this in app-template by shipping some JSON file to configure RA.

I guess the other way to make this work is to make `probe-run` ignore these extra flags if it's being used to run tests but that would be kind of hacky.

Lotterleben, 2020-12-14:
Ah! Thank you for the detailed explanation.
To me, the hacky fix actually seems nicer from the outside: If you don't look under the bonnet, things *just work* (no "put these JSON files there to make this work", which is part of the rust ecosystem magic for me) plus we don't have to wait for R-A to maybe introduce test flag configuration someday.

closed 2021-02-15 by bors[bot]