knurling-rs/defmt #271: list of not directly user-facing breaking changes for 0.2.0

japaric, 2020-11-24:
(list not final)

# `write!` tag

background: PR #263 
what: add a new tag to lift the restriction of 'only a single `write!` call is allowed in an `impl Format` block'

# `assert_eq!` tag

background: PRs #266 and probe-run#110
what: add a new tag to the log statement used in `defmt::assert_eq!` so that printers can directly format defmt data into a color diff
note: current probe-run parses formatted output and re-formats it to add the color diff

japaric, 2020-11-30:
> write! tag

this may not be necessary if we take the Formatter by value (?) #277

jonas-schievink, 2020-12-01:
We still might want that to distinguish `write!` from derive-generated strings, for structured output

jonas-schievink, 2021-02-09:
Triage: `write!` now has its own tag, `assert_eq!` does not

Urhengulas, 2021-02-28:
Hi, should we put this list into the documentation or changelog? Else this can be closed I guess?

Urhengulas, 2021-03-01:
Implementing `assert_eq!` will be very similar to `write!` and therefore rather straight-forward.

Urhengulas, 2021-05-12:
Maybe it is obvious, but what is the purpose of this list?

japaric, 2021-06-23:
> Maybe it is obvious, but what is the purpose of this list?

it's a bit like the breaking-change label except that the list only contains refactors / cleanups (no new features, bug fixes or anything the user will notice)

---

>  assert_eq! tag 

looking into this: currently the re-parsing + coloring happens in
https://github.com/knurling-rs/defmt/blob/fde2aab0bc541081d7def42fb049a36fa793a2bc/decoder/src/log.rs#L163

this is too late (too deep in the call stack) because there's no access to the decoded defmt frame (e.g. its tags or the log level) at this point.

A better place would be around here: 
https://github.com/knurling-rs/defmt/blob/fde2aab0bc541081d7def42fb049a36fa793a2bc/decoder/src/log.rs#L43

Here the defmt `Frame` is available. At this point the `Frame` gets converted into something printable (`impl fmt::Display`). We could either:
- add a `AssertEq` variant to `Level`. A bit hacky but the `assert_eq!` log message will always be printed at the error log level
- add a general purpose `tags` field (a `Vec` maybe) to `Frame`. If it contains the `assert_eq` tag then colorize the diff and print it at the error log level (ignore the `level` field in `Frame`)