knurling-rs/defmt #286: Git DEFMT_VERSION doesn't get updated when doing a Git commit.

Dirbaio, 2020-11-27:
- In a firmware project, add a `[patch.crates-io]` pointing to a local checkout of the defmt repo
- Modify some defmt `.rs` files
- Build the firmware, it gets `DEFMT_VERSION = <commit 1>`. 
- The firmware runs great with the modified defmt, you decide to do a defmt commit to prepare a juicy PR
- do `git commit` in the defmt repo
- Now defmt HEAD is `<commit 2>`
- Build the firmware, it STILL gets `DEFMT_VERSION = <commit 1>` because build.rs doesn't re-run because you didn't edit any .rs files since last build.

japaric, 2020-11-30:
This git mechanism is meant to be used with *git* dependencies.

> In a firmware project, add a [patch.crates-io] pointing to a local checkout of the defmt repo

is the override a path dependency or a git dependency? it should be a git dependency

``` toml
[patch.crates-io]
defmt = { git = "file:///home/japaric/rust/defmt", branch = "wip-branch" }
#                ^^^^ git dependency that lives in the local file system
```

with a git dependency after you do a defmt commit you need to run `cargo update` from the firmware folder to make the lock file use the newer commit

---

the problem with path dependencies is that:

1. git commits are not recognized by Cargo as changes so, afaik, there's no way to re-run `build.rs` on a new commit. the work around is to make the build script *always* run by doing `touch build.rs` from it but this causes crates to be re-built every time you do `cargo build` (even if nothing changed)

2. you can build a crate that has uncommited changes; these changes won't be included in the `DEFMT_VERSION` string -- the string will still report the last commit hash. there are some workarounds for this: you can include a hash of `git status` in `DEFMT_VERSION` but not sure if that's too reliable because of (1).

jonas-schievink, 2021-01-11:
> 1. git commits are not recognized by Cargo as changes so, afaik, there's no way to re-run `build.rs` on a new commit.

Super hacky idea, but would it work to emit rerun-if-changed directives for `.git/HEAD` and whatever file `HEAD` points to? Not terribly familiar with git, but I *think* that should cover new commits?