knurling-rs/defmt #287: Separate "crate version" from "wire format version"

Dirbaio, 2020-11-27:
Currently the `defmt` crate version is embedded in the ELF, and is used by the decode to check compatibility of the wire format. The version is the crates.io version or the Git commit hash if building from a git clone.

Probe-run refuses to run when the version doesn't match exactly. You have to clone it, patch it to use the right defmt version, rebuild it. This is quite painful in these scenarios:

- When developing defmt, trying to run a firmware using the WIP modified version.
- When releasing a lib / project that depends on git defmt, for people trying to use it.

Many (most?) changes don't break wire format. Even API-breaking changes like #258 can not-break the wire format. 

Why not separate the "wire format version" concept from the "crate version"?

- `defmt` crate has a WIRE_FORMAT_VERSION const. 
- `defmt-decoder` crate has a WIRE_FORMAT_VERSION const. 
- `defmt-decoder` checks the WIRE_FORMAT_VERSION from the ELF matches its const, fails if not.
- the comparison can be a simple string equality, no need for semver stuff I think.
- When doing wire-format breaking changes, the wire format version gets bumped

Advantages:

- No need for build.rs magic at all.
- Easier hacking, easier use of git defmt versions that are still compatible with released wire format version
- possibility of doing a major version `defmt` crate release that doesn't bump the wire format version. (ie with API-breaking changes but not with wire-format-breaking changes)

bradleyharden, 2020-11-27:
This was a pain for me when trying to use a branch from @jacobrosenthal to get ATSAMD51 support. I would appreciate it.

japaric, 2020-11-30:
disadvantages / not advantages (?)

1. the `main` branch still need versioning, e.g. `git.2-pre.1`, `git.2-pre.2` for each breaking change preceding the v0.2.0 release

2. either the PR author or the reviewer needs to correctly identify the breaking change and manually bump the version. Failure to do so = runtime errors for git users. It's not straightforward to recognize breaking changes: e.g. to my surprise extending the format string syntax is a breaking change.

3. harder to identify which defmt commit hash you can use with your git version of your `probe-run` which would now report `git.2-pre.1` instead of a commit hash

4. even if you identify the commit range of defmt that's compatible with your git version of probe-run you still will want to use the exact commit hash of the decoder / elftable dependency used by probe-run because there may have been API breaking changes there that are not reflected in the 'wire version' -- unless we also track those in the 'wire version' which means more work in (2)

---

> possibility of doing a major version defmt crate release that doesn't bump the wire format version. (ie with API-breaking changes but not with wire-format-breaking changes)

note that we can still do this with the current scheme for crates.io releases: we can do 'if minor = 3 set version to 0.2' in the build script.

commit c7b20d53 (2021-07-27): Merge #540

540: Separate "crate version" from "wire format version"  r=jonas-schievink a=Dirbaio

Depends on #539. Fixes #287.

This decouples the "wire format version" from the crate version (Cargo package version or git commit). Wire format version can now be bumped independently. This alleviates pains listed in #539., and more importantly it will allow making breaking changes to the wire format in a semver-minor `defmt` crate release. See #426.

Version is `3` to be able to retroactively number `0.1.x = 1, 0.2.x = 2`. A followup PR will add support to `defmt-decoder` to decode both version [... cut]
PR #540 (2021-07-27): Separate "crate version" from "wire format version" 
Depends on #539. Fixes #287.

This decouples the "wire format version" from the crate version (Cargo package version or git commit). Wire format version can now be bumped independently. This alleviates pains listed in #539., and more importantly it will allow making breaking changes to the wire format in a semver-minor `defmt` crate release. See #426.

Version is `3` to be able to retroactively number `0.1.x = 1, 0.2.x = 2`. A followup PR will add support to `defmt-decoder` to decode both version 2 and 3 concurrently.
closed 2021-07-27 by bors[bot]