knurling-rs/defmt #292: support impl Debug and impl Display values directly in logging macros

japaric, 2020-11-30:
This is different from the existing adapter types: Debug2Format and Display2Format in that formatted string will never be truncated. It still has the disadvantage of "no compression"

the rough idea is to
1. extend defmt wire format to support "unsized values" (streams?). That is values whose length (in bytes once encoded) is not known before the value's data written into a `Formatter`
2. extend the format string with say `{:Debug}`. This string expects an `impl Debug` value. The "unsized" value is then written into the `defmt::Formatter` probably using some adapter to make `defmt::Formatter` look like a `impl core::fmt::Write` type.

jonas-schievink, 2021-01-12:
Since the result of `Display` and `Debug` is always valid UTF-8, we could use an invalid UTF-8 byte as the end-of-frame delimiter (say, `0xFF`). That way we don't have to implement something more complex or expensive like SLIP or even COBS.

jonas-schievink, 2021-01-12:
These will unfortunately make our format string syntax deviate quite a bit from `core::fmt`, since these won't be able to use `{}` and `{:?}` like you would expect from `core::fmt`.

Could we maybe force these to go through `{:?}`/`Format` and use some secret syntax that isn't exposed to the user? Then `Display2Format` and `Debug2Format` would remain, but become simple wrappers.

closed 2021-01-13 by bors[bot]