knurling-rs/defmt #298: Add CI tests for defmt-test

jonas-schievink, 2020-12-03:
*Who tests the test-runner?*

Sh3Rm4n, 2020-12-09:
If I understand it correctly, this about two possible three test steps:

1. Check that the generated code is buildable
2. Introduce proc-macro unit tests which check generated output against expected output
3. Check if the code can be run (which is difficult for `thumb*` targets with the CI)

If I have looked up correctly, no CI job is run for  `defmt-test` at the moment, so `1.` could be implemented by introducing a CI job.

japaric, 2020-12-09:
now that `defmt-test` is in the defmt repository it should be possible to run firmware that uses defmt-test in the emulator using `qemu-run` (or at least I hope that works).

We should check two scenarios: all defmt-test tests pass and one defmt-test test fails; we have firmware tests that panic [here]. We should also check that the output of defmt-test matches our expectations -- we use [snapshot testing] for that.

[here]: https://github.com/knurling-rs/defmt/blob/c599ba26626aeae98fb97a2cae71b3da333411e1/firmware/qemu/src/bin/panic.rs
[snapshot testing]: https://github.com/knurling-rs/defmt/pull/222

japaric, 2020-12-09:
> it should be possible to run firmware that uses defmt-test in the emulator using qemu-run (or at least I hope that works).

those firmware tests / examples live in the [firmware/qemu](https://github.com/knurling-rs/defmt/tree/c599ba26626aeae98fb97a2cae71b3da333411e1/firmware/qemu) folder.

Adding a new program that uses defmt-test to `firmware/qemu/src/bin` should cover your point (1).
Listing that program in [this script](https://github.com/knurling-rs/defmt/blob/c599ba26626aeae98fb97a2cae71b3da333411e1/firmware/qemu/test.sh) should cover your points (3)

In general, we don't test proc-macro expansion against expected Rust code (your point (2)) in this repo

PR #299 (2020-12-10): Test defmt-test
Add defmt-test executed by qemu-run, as suggested in https://github.com/knurling-rs/defmt/issues/298#issuecomment-741848194. 

Closes #298
closed 2020-12-10 by japaric