knurling-rs/defmt #317: version mdbook

Lotterleben, 2020-12-18:
problem:
`main` contains breaking changes (like https://github.com/knurling-rs/defmt/pull/313) that v0.1.x, i.e. the current crates.io version does not have or does differently. We need a way to give both crates.io- as well as github users docs by versioning the book like rtic.rs does for example

japaric, 2020-12-18:
update:
for now we are using two domains `defmt.` for v0.1.x and `defmt-next.` for v0.2.0.
we may want to host (all?) minor versions under the same subdomain as rtic.rs does

Lotterleben, 2020-12-21:
I think that closes this issue for now.
imo I'd only start hosting hosting all minor versions when there's explicit demand for it– but that's speaking as someone who keeps landing on (outdated) versioned docs though wild google searches, which gets really confusing if you don't notice straight away.

closed 2020-12-21 by Lotterleben