knurling-rs/defmt #319: String fields in structs should be formatted using ':?' hint

japaric, 2020-12-18:
Current output:

``` rust
#[derive(Format)]
struct S { x: &'static str }

info!("{}", S { x: "hi" });
// -> INFO S { x: hi }
```

Desired output: `S  { x: hi }`

---

I wasn't sure how to best resolve this one.
In the RFC, we discussed applying the ':?' hint to all struct fields and that does the job in the previous example.
But seems like it would have the adverse effect of display hints never propagating into structs. Example below:

``` rust
// #[derive(Format)]
struct S { x: &'static str, y: u8 }

// derive would expand into
impl defmt::Format for S {
    fn format(&self, f: defmt::Formatter) {
        write!(f, "S {{ x: {=str:?}, y: {=u8:?} }}")
    }
}
```

``` rust
// matches the expected output
info!("{}", S { x: "hi", y: 42 });
// -> INFO S { x: "hi", y: 42 }

// `y` field ignores top-level `:x` display hint
info!("{:x}", S { x: "hi", y: 42 });
// -> INFO S { x: "hi", y: 42 }
```

The second statement ignores the ':x' hint.

We discussed in the RFC that the innermost display hint should not be overridden by an outer one.
Example from the RFC:

``` rust
struct S { x: u8 }
impl Format for S {
    fn format(&self, f: defmt::Formatter) {
        write!(f, "S {{ x: {=u8:x} }}", self.x)
    }
}

// binary hint should be ignored
info!("{:b}", S { x: 42 });
// -> INFO S { x: 0x2a }
```

Given that, using ':?' in fields would make non-string fields ignore display hints specified at the top call which is not what I would expect.

japaric, 2021-01-05:
Here's one way to make this work.

> We discussed in the RFC that the innermost display hint should not be overridden by an outer one.

We revise this part of the specification to say "innermost display hint should not be overridden by an **applicable** outer one" and introduce the idea of certain hints not being applicable to certain types.

For example, `:x`, `:b` and `:a` do *not* apply to type `=str`, but `:?` does (it escapes the string).
Similarly, `:?` does *not* apply to integer types like `=u8` but `:b` and `:x` do apply.

With that adjusted semantics we can tweak the derive macro to always add the `:?` hint to all struct / enum fields. Example:

``` rust
#[derive(Format)]
struct S { x: &'static str, y: u8 }

// derive would expand into
impl defmt::Format for S {
    fn format(&self, f: defmt::Formatter) {
        write!(f, "S {{ x: {=str:?}, y: {=u8:?} }}")
    }
}
```

Now if the user does:

``` rust
info!("{:x}", S { x: "hi", y: 42 });
```

A printer (like `probe-run`) needs to consider all the nested hints and apply the innermost *applicable* one. Following the previous example:

- field `x` has hints `:?` and `:x` (in the innermost to outermost order). `x` has type `&str` and `:?` is applicable to `&str` so `x` is formatted with the `:?` hint.

- likewise, field `y` has hints `:?` and `:x`. `y` has type `u8` so the outermost hint `:?` does *not* apply. Therefore, `y` is formatted with the `:x` hint.

The output then is `S { x: "hi", y: 0x2a }`

Looking at the other example from the issue description:

``` rust
struct S { x: u8 }
impl Format for S {
    fn format(&self, f: defmt::Formatter) {
        write!(f, "S {{ x: {=u8:x} }}", self.x)
    }
}

// binary hint should be ignored
info!("{:b}", S { x: 42 });
```

Here field `x` has hints `:x` and `:b` (in the innermost to outermost order). `x` has type `=u8` and hint `:x` applies to that type so hint `:b` is ignored (as expected).

Urhengulas, 2021-07-06:
@japaric Do you consider this fixed by #359?

japaric, 2021-07-06:
yes!

closed 2021-07-06 by japaric