knurling-rs/defmt #320: Unable to print dates and times

David-OConnor, 2020-12-19:
This example is using the `chrono` trait, but its' a broader issue I can't identify. This is 

With `rtt`, you could do this:
```rust
defmt::info!("DEBUG date: {:?}", &rtc.get_datetime().unwrap());
```

When I try with `defmt`:
```rust
    |     defmt::info!("DEBUG date: {:?}", &rtc.get_datetime().unwrap());
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `Format` is not implemented for `NaiveDateTime`
```

Any ideas? Thank you.

Sh3Rm4n, 2020-12-19:
This is because defmt relies on a custom `Format` trait, which does not use the implementations provided by the rust core traits `Debug` and `Display`. 

On custom types you automatically implement the `Format` trait via `#[derive(Format)]` . 
As `NaiveDateTime` is a type provided by `chrono` you can not add this impl directly to it. 

A way to resolve this could be to use the `NewType` Pattern to implement Format or get a meaningful primitive type out of `NativeDateTime` like `u32` etc., which is supported by defmt. 

There is probably also a better way I can't think of as of now.

MathiasKoch, 2020-12-19:
The easier way of doing what @Sh3Rm4n proposed, is available in the `defmt::Debug2Format` & `defmt::Display2Format` here: https://github.com/knurling-rs/defmt/blob/main/src/lib.rs#L653-L687

David-OConnor, 2020-12-20:
Thank you - I'll check those out.

jonas-schievink, 2020-12-21:
This is behaving as expected, but marking as documentation improvement just in case we can do better there

David-OConnor, 2021-01-10:
I think another option is make it work the way `rprintln` does.

japaric, 2021-01-11:
> Is it unfeasable to make it handle this (perhaps as a fallback?)

currently, yes. at least until 'unsized values' are supported in defmt (see #292).

> Or is not using Debug and Format core to how this works?

no, defmt does deferred formatting so it doesn't use `core::Debug` or `core::fmt`. (if you are interested in reading how defmt does deferred formatting you can check out the [design / impl docs](https://defmt.ferrous-systems.com/design.html)).

commit ba466d07 (2021-03-26): Merge #437

437: `book`: Document adapter types `Display2Format`, `Debug2Format` r=jonas-schievink a=Urhengulas

Documents the workaround for and thereby fixes #320
PR #437 (2021-03-26): `book`: Document adapter types `Display2Format`, `Debug2Format`
Documents the workaround for and thereby fixes #320
closed 2021-03-26 by bors[bot]