knurling-rs/defmt #322: Trouble printing in interrupts

David-OConnor, 2020-12-20:
Code:

```rust
#[interrupt]
fn EXTI4() {
    defmt::warn!("DN");

}
```

Program crashes once this interrupt fires:
```rust
  (HOST) ERROR failed to decode defmt data: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]  // Sometimes many more 0s, and sometimes non-0 numbers.
```
For context, `rprintln` works in this case.

Sh3Rm4n, 2020-12-20:
Does this also happen when you put this statement in the main loop like so?

```rust
fn main() -> ! {
    // initailization
    loop {
        defmt::warn!("DN:);
    }
}
```

Does this stop happening, when you reset the interrupt flag inside the interrupt, so it does not fire indefinitely?

David-OConnor, 2020-12-20:
Nope; works in that case.

jonas-schievink, 2020-12-21:
Can you post the whole code needed to reproduce this? This should definitely work – the defmt sink acquires a lock internally, so usage in the presence of preemption should not be a problem.

Sh3Rm4n, 2020-12-30:
I have tried to reproduce this issue, but was not successful.

Example, based on the app-template:

```rust
#![no_std]
#![no_main]

use app_template as _;
use stm32f3xx_hal::{pac, interrupt};
use cortex_m::{asm, interrupt::free};

#[cortex_m_rt::entry]
fn main() -> ! {
    defmt::info!("Initalization");
    let dp = defmt::unwrap!(pac::Peripherals::take());
    dp.EXTI.imr1.write(|w| w.mr4().set_bit());
    unsafe {
        cortex_m::peripheral::NVIC::unmask(pac::Interrupt::EXTI4);
    };

    // trigger interrupt
    dp.EXTI.swier1.write(|w| w.swier4().set_bit());

    defmt::info!("Main Loop:");
    loop {
        asm::delay(1_000_000);
        defmt::info!("Trigger Itr");
        dp.EXTI.swier1.write(|w| w.swier4().set_bit());
    }
}

#[interrupt]
fn EXTI4() {
    free(|_| {
        // Clear the interrupt flag, to prevent continous firing.
        unsafe { (*pac::EXTI::ptr()).pr1.modify(|_, w| w.pr4().set_bit()) }
    });

    defmt::warn!("DN");
}
```

Even without delay and interrupt clearing, this is working as expected. 
What could be missing @David-OConnor?

David-OConnor, 2021-01-02:
Hi. I'm not sure how to repro this. I was wrong - I don't think it's related to interrupts. I still run into this frequently, but have yet to track down what's going on.

David-OConnor, 2021-01-10:
Of note, I've been able to reproduce this on a custom F303 board, a F3 Discovery board, and an L4 nucleo board.

japaric, 2021-01-11:
> One bit of info: Sometimes wiping the memory using ST-Link utility helps.

are you, perhaps, using `probe-run`'s `--no-flash` flag? that flag is not meant to be used together with defmt logs.

> I think this is wfi related.

does using `probe-run`'s `connect-under-reset` flag help in any way?

David-OConnor, 2021-01-11:
No - I'm using the `app-template` defaults.

I'll check `connect-under-reset`.

Urhengulas, 2021-03-14:
Hi @David-OConnor,

I think your issue is the same or at least similar to https://github.com/knurling-rs/probe-run/issues/170. I am currently trying to reproduce it and would highly appreciate your help!

Let's start easy: What version of `defmt` and `probe-run` did you first encounter this issue (git or crates-io?)?

David-OConnor, 2021-03-15:
I've been getting this for a few months. Currently get this with probe-run `0.2.1`, and defmt version `0.2`. Don't know the original version.

chenyanming, 2021-03-16:
Same here. Very easy to get `(HOST) ERROR failed to decode defmt data` when printing `&[u8]` slice in 0.2+ version. Once I back to 0.1, it works normally.

David-OConnor, 2021-04-03:
Hi - this is caused by a limitation of at least some (most?) stm32 families. Workaround:

```rust
#[cfg(not(any(feature = "g0", feature = "h7")))]
/// Workaround due to debugger disconnecting in WFI (and low-power) modes.
/// This affects most (all?) STM32 devices. In production on battery-powered
/// devices that don't use DMA, consider removing this, to prevent power
/// use by the DMA clock.
/// For example, see STM32F446 errata, section 2.1.1.
pub fn debug_workaround(dbgmcu: &mut pac::DBGMCU, rcc: &mut pac::RCC) {
    #[cfg(not(feature = "l5"))]
    dbgmcu.cr.modify(|_, w| w.dbg_sleep().set_bit());
    dbgmcu.cr.modify(|_, w| w.dbg_stop().set_bit());
    dbgmcu.cr.modify(|_, w| w.dbg_standby().set_bit());

    #[cfg(feature = "f3")]
    rcc.ahbenr.modify(|_, w| w.dma1en().set_bit());

    #[cfg(not(feature = "f3"))]
    rcc.ahb1enr.modify(|_, w| w.dma1en().set_bit());
}
```

See [this errata](https://www.st.com/content/ccc/resource/technical/document/errata_sheet/c3/6b/f8/32/fc/01/48/6e/DM00155929.pdf/files/DM00155929.pdf/jcr:content/translations/en.DM00155929.pdf#%5B%7B%22num%22%3A37%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C67%2C724%2Cnull%5D)

closed 2021-04-03 by David-OConnor

Urhengulas, 2021-04-05:
Hi @David-OConnor, Props for tracking this bug down!

I'd put this into a "limitations" section. What do you think?

David-OConnor, 2021-04-05:
Sounds good