knurling-rs/defmt #330: Examples on custom implementation of the `Format` trait.

mexus, 2021-01-08:
Currently it is virtually impossible to have a custom implementation of the `defmt::Format` trait when the corresponding derive macros is inappropriate: the `defmt::Formatter` has all its methods hidden from the docs, but even if they were visible, it is completely unobvious how to call them correctly.

I've tried to use a different approach by utilizing the `defmt::write!` macro, but it didn't help either: for example, the following program outputs `10 khz` and panics for a reason unknown.
```rust
struct A {
    field: u8,
}

impl Format for A {
    fn format(&self, fmt: &mut defmt::Formatter) {
        defmt::write!(fmt, "{:?} khz", self.field);
        defmt::write!(fmt, "{:?} khz", self.field);
    }
}

#[cortex_m_rt::entry]
fn main() -> ! {
    defmt::info!("{:?}", A { field: 10 });

    light_organ::exit()
}
```

japaric, 2021-01-11:
Hello @mexus. Have you come across [this part of the documentation](https://defmt.ferrous-systems.com/format.html#write)? If not, then we should make it more visible somehow, maybe linking to it from the API docs.

> the defmt::Formatter has all its methods hidden from the docs

those are unstable implementation details not meant to be consumed by users of defmt

> utilizing the defmt::write! macro,

currently, that (and calling other type's `Format::format` method) is the only way to write a custom `Format` implementation

> the following program outputs 10 khz and panics for a reason unknown.

that's expected behavior: `write!` cannot be called more than once from a `format` implementation (this operation will be prevented at compile time in defmt v0.2.0)

mexus, 2021-01-11:
Hi @japaric and thanks for a detailed reply!

> Have you come across [this part of the documentation](https://defmt.ferrous-systems.com/format.html#write)?

Yes, I happened to, but putting a link at the API docs sounds great :)

> that's expected behavior: `write!` cannot be called more than once from a `format` implementation (this operation will be prevented at compile time in defmt v0.2.0)

OMG I've somehow missed the NOTE on that :(

Well it makes thing a lot more complicated, although I think the limitation can be bypassed by creating wrapper types for the amount of `write!` calls required.

Thanks a lot!

closed 2021-01-12 by bors[bot]