knurling-rs/defmt #336: Running ui tests on mac

BriocheBerlin, 2021-01-12:
When trying to get started to write compile fail tests , running
`$ cargo test --test ui`
didn't work for me using a mac. I'd always get this unhelpful error:
```
LLVM ERROR: Global variable '{"package":"defmt","tag":"defmt_prim","data":"{=i8}","disambiguator":"2773326613612971761"}' has an invalid section specifier '.defmt.prim.{"package":"defmt","tag":"defmt_prim","data":"{=i8}","disambiguator":"2773326613612971761"}': mach-o section specifier requires a segment whose length is between 1 and 16 characters.
error: could not compile `defmt` 
```

Passing more flags like this did help:
`$ cargo test -p defmt --all-features --test ui`

This makes sense when looking at [ci.yml](https://github.com/knurling-rs/defmt/blob/main/.github/workflows/ci.yml#L50)

japaric, 2021-01-12:
@BriocheBerlin could you test this on macOS? I'd like to figure out if `defmt` fails to compile natively or if this problem is due to the single workspace configuration used in this repository.

``` console
$ cargo new --bin hello
$ cd hello
$ # or manually add 'defmt = "0.1.3"' to Cargo.toml
$ cargo add defmt 
$ # does this succeed?
$ cargo run
```

BriocheBerlin, 2021-01-12:
`$ cargo run` succeeded.

japaric, 2021-01-12:
@BriocheBerlin could you run `cargo test --test ui -j1` in this repo and report which crate fails to build? That command builds one crate at a time rather than in parallel.

BriocheBerlin, 2021-01-13:
So it seems that in the working version on the `hello` project the command for building the defmt crate is very similar. However it uses a different `src/lib.rs` file which is local `<more_path>/.cargo/registry/src/github.com-1ecc6299db9ec823/defmt-0.1.3/src/lib.rs`.

japaric, 2021-01-14:
hmm, maybe the macOS issue was introduced recently so it's in the git version but not on crates.io?
could you try these two:
``` console
$ cargo new --bin hello
$ cd hello

$ # or add 'defmt = { git = "https://github.com/knurling-rs/defmt" }' to Cargo.toml dependencies
$ cargo add defmt --git https://github.com/knurling-rs/defmt

$ # does this succeed?
$ cargo run
```

``` console
$ git clone https://github.com/knurling-rs/defmt
$ cd defmt
$ git checkout defmt-v0.1.3

$ # do these produce the llvm error?
$ cargo build
$ cargo test
```

BriocheBerlin, 2021-01-14:
`defmt-v0.1.3` works without problems. However the `hello` thing produces this interesting error:

japaric, 2021-01-14:
oh, yes. my bad it should be
``` toml
defmt = { git = "https://github.com/knurling-rs/defmt", branch = "main" }
```
in the Cargo.toml

BriocheBerlin, 2021-01-14:
Ah, doesn't work. So the resulting commands are slightly different.
Working version (using `defmt = "0.1.3"` in Cargo.toml):
```
rustc --crate-name defmt 
--edition=2018 
<more_path>/.cargo/registry/src/github.com-1ecc6299db9ec823/defmt-0.1.3/src/lib.rs 
--error-format=json 
--json=diagnostic-rendered-ansi 
--crate-type lib 
--emit=dep-info,metadata,link 
-C embed-bitcode=no 
-C debuginfo=2 
-C metadata=4f3b6bbbb29948b6 
-C extra-filename=-4f3b6bbbb29948b6 
--out-dir <more_path>/hello/target/debug/deps 
-L dependency=<more_path>/hello/target/debug/deps 
--extern defmt_macros=<more_path>/hello/target/debug/deps/libdefmt_macros-f658b9f3b197fd12.dylib 
--extern heapless=<more_path>/hello/target/debug/deps/libheapless-9846ae8300a43c3e.rmeta 
--cap-lints allow 
-L <more_path>/hello/target/debug/build/defmt-d1e4ca37c018f197/out
```
With `defmt = { git = "https://github.com/knurling-rs/defmt", branch = "main" }` it doesn't have
`--extern heapless=<path>`
When running on defmt as is, this line is different:
`-L dependency=<more_path>/hello/target/debug/deps `
Instead I got:
`-C incremental=<more_path>/defmt/target/debug/incremental `

japaric, 2021-01-14:
Then this more serious: as things are `defmt` 0.2 won't compile on macOS which is going to hinder adoption: using defmt in a no-std, non-embedded crate would make the crate unusable on macOS.

I'll put this on the list of blockers for 0.2.0. cc @jonas-schievink

jonas-schievink, 2021-01-14:
The invalid section names were already used in defmt 0.1 though? Or is there another error?

japaric, 2021-01-14:
I think this may have been caused by PR #291 . After the changes in the macros we emit `link_section` on x86_64 macOS -- before we emitted `fetch_add_string_index` on x86_64 macOS.

BriocheBerlin, 2021-01-14:
That would explain why in the working `src/lib.rs` file it was always `target_arch = "x86_64"` instead of `feature = "unstable-test"`.

BriocheBerlin, 2021-01-14:
The last commit where `$ cargo build` works for me is https://github.com/knurling-rs/defmt/commit/04cf5731ecd35a3c17deb876b6be224194ca9b6a. 
(But `$ cargo test` fails, again complaining about missin `InternalFormatter`)/

japaric, 2021-01-15:
we looked into this. the llvm error can be fixed by changing the linker section names we use.
currently we use `.defmt.print.<LONG JSON STRING HERE>`
on macOS the longest linker section name can be `.defmt,0123456789ABCDEF` (note **comma** after `.defmt` not period)
we should adjust the macros to generate a linker section name like that
instead of <JSON STRING> we should use digest / hash of that and truncate it to not exceed the 16 character limits

japaric, 2021-01-15:
for starters this proc-macro needs to be changed (but probably others also need to be changed to work on macOS):
https://github.com/knurling-rs/defmt/blob/eb8c3b102dc2872d7051257df37e336011a8e6fe/macros/src/lib.rs#L987

japaric, 2021-01-15:
we should also build and run the following std program on linux, macos and windows to shake out any other linker section name related bugs:
``` rust
// Cargo.toml uses defmt via a path dependency

fn main() {
    defmt::info!("hello");
}

#[defmt::global_logger]
struct Logger;

unsafe impl defmt::Logger for Logger {
    fn acquire() -> Option<core::ptr::NonNull<dyn defmt::Write>> {
        None
    }
    unsafe fn release(writer: core::ptr::NonNull<dyn defmt::Write>) {}
}

defmt::timestamp!("{=u32}", 0);
```

Urhengulas, 2021-02-02:
This issue seems resolved, right?

jonas-schievink, 2021-02-02:
Yup!

closed 2021-02-02 by jonas-schievink