knurling-rs/defmt #348: implement `Format` for `core::time::Duration`

japaric, 2021-01-19:
it would be nice if this type was supported

David-OConnor, 2021-01-19:
Same with all `chrono`, or `rtcc` types.

jonas-schievink, 2021-01-19:
External crates need to implement `Format` by themselves, like they have to do for serde traits. Implementations for libcore/liballoc types should be added here though.

jonas-schievink, 2021-02-03:
This would be easier to implement if we could pad a value with zeroes (`{:09}` syntax)

Urhengulas, 2021-02-13:
I was just wondering if it is possible to apply own macros to foreign types, similar to how it is possible to impl own traits on foreign types?
This would be very handy to easily implement `Format` for the standart types.

Urhengulas, 2021-02-13:
Also it doesn't seem to be possible to implement it like following for `Duration`. Can someone quickly explain why not?

```rust
// src/impls.rs
impl Format for core::time::Duration {
    fn format(&self, fmt: Formatter) {
        crate::write!(
            fmt,
            "Duration {{ {=u64} {=u32} }}",
            self.as_secs(),
            self.subsec_nanos(),
        )
    }
}

// Error: failed to resolve: could not find `defmt` in `{{root}}`

jonas-schievink, 2021-02-15:
> failed to resolve: could not find `defmt` in `{{root}}` 

This is likely because the macros expect the invoking crate to depend on `defmt`, and thus have that name available. You should be able to `use crate as defmt;` in the crate root to fix this.

Urhengulas, 2021-02-15:
@jonas-schievink I've tried that, but still doesn't work. I've also tried to replace `crate::write!` with `defmt_macros::write!`, but still didn't work.

May you have a look here? https://github.com/knurling-rs/defmt/blob/348-format-for-duration/src/impls.rs#L462-L471

Urhengulas, 2021-02-15:
Okay, I fixed it for my case, but I am unsure if this breaks other stuff. Let's see in the reviews 😺

commit 42be5f61 (2021-02-16): Merge #391

391: 348 format for duration r=jonas-schievink a=Urhengulas

This implements the `trait defmt::Format for core::time::Duration` and also adapts `defmt_macros::write!` to achieve this.

Fixes #348
PR #391 (2021-02-16): 348 format for duration
This implements the `trait defmt::Format for core::time::Duration` and also adapts `defmt_macros::write!` to achieve this.

Fixes #348
closed 2021-02-16 by bors[bot]