knurling-rs/defmt #366: turn CI logic into xtasks

japaric, 2021-02-03:
would be nice if there were xtasks to run the CI tests locally.
right now CI logic is in a yaml file that contains bash-like stuff and the tests.sh bash file (which won't run on Windows)
Ideally I would like the GHA yaml be more or less just a call of different xtasks
``` yaml
# apt-get install stuff has been omitted
jobs:
  host:
    # omitted: OS matrix
    name: run unit tests on the host
    run: cargo xtask test host

  cross:
    name: check cross compilation
    run: cargo xtask test cross

  snapshot:
    name: run snapshot tests (QEMU)
    run: cargo xtask test snapshot

  book:
    name: run mdbook tests
    run: cargo xtask test book

  lint:
    name: run rustfmt and clippy
    run: cargo xtask test lint
```

and then a single xtask, e.g. `cargo xtask test all`, that sort of runs all the above jobs which is what contributors would run locally to check their branch / PR is "green".
the xtasks should auto-install the cross compilation targets (`rustup target add`) so that contributors don't need to do "setup" steps
(maybe the xtasks should uninstall the cross compilation targets that were not there when they exit, regardless of the exit code)

Urhengulas, 2021-02-03:
Another idea to run the test locally would be [act](https://github.com/nektos/act), which runs the github actions inside a docker container.

jonas-schievink, 2021-02-03:
We could also move the OS-specific logic into xtask and then only expose `cargo xtask test`, or even `cargo test -p xtask` to run everything

PR #393 (2021-04-08): implement CI tests as xtask
this xtask implementation fixes #366 and
- collects all test errors in one grand result vec and displays them at the end
- has commands for `test-all`, `test-host`, `test-cross` etc.
- has `-k` switch to keep targets that were installed as required dependency
- displays colored diff for failed snapshot tests
- puts "macos book build fixes" in separate commit ( c847772 ) - please review
- updates the ci yaml to use this xtask

not done:
- install os specific packages like `libusb` (out of scope, IMO)
closed 2021-04-08 by bors[bot]