knurling-rs/defmt #370: Ability to format with a specific radix.

newAM, 2021-02-08:
Hey there, love the crate!

One feature I would like to have is the ability to format a number with a specific radix.

I have read the [Single Format trait](https://defmt.ferrous-systems.com/design-format.html?highlight=hex#single-format-trait) section of the book:
> defmt does not have all these [radix] formatting traits. The rationale is that the device should not make the decision about how an integer is formatted. The formatting is done in the host so the host should pick the format.

This makes sense for majority of cases, however there are some cases were you **really** want a specific format.

## Usecase

I came across this while developing [w5500-ll](https://github.com/newAM/w5500-ll-rs), a driver for an embedded networking chip, and there are two cases where it would be very helpful to ensure data is represented in a specific radix:

* An IPv4 address should be represented as `"{}.{}.{}.{}"` with the bytes in base 10.
* An EUI-48 MAC address should be represented as `"{:02X}:{:02X}:{:02X}:{:02X}:{:02X}:{:02X}"` with the bytes in base 16.

Urhengulas, 2021-02-08:
Hi @newAM,
Thank you for filing your feature request 😄 

Could you please elaborate why it is helpful to format the data on the device? Do you see alternatives to formatting a number with a specific radix?

jonas-schievink, 2021-02-08:
The git version of defmt already supports `{:x}`, `{:X}`, and `{:b}` (https://defmt-next.ferrous-systems.com/hints.html). I'd accept a PR that adds support for octal numbers, matching `core::fmt::Octal`, and support for padding is also something I'd love to see (both whitespace padding and sign-aware zero padding with `:0N`).

I don't think we need to support *arbitrary* radices though (like `uNN::from_str_radix`), since there's no counterpart in `core::fmt`, and the feature feels pretty fringe.

> Could you please elaborate why it is helpful to format the data on the device?

I don't think they're asking for that – if you need this defmt isn't the right tool. I interpreted this to be about having the device choose how the host should format a number, which is the direction we've been moving towards in the git version.

newAM, 2021-02-08:
> The git version of defmt already supports {:x}, {:X}, and {:b}

I missed checking the unreleased version, this is exactly what I am looking for!
I'll make sure to check this next time.

Thanks for all the hard work with this crate! :+1:

closed 2021-02-08 by newAM