knurling-rs/defmt #381: defmt-test: support the #[ignore] attribute & raise error on unknown attributes

japaric, 2021-02-10:
The standard test runner uses this attribute to ignore (omit running) a unit test. We implement the same functionality or, at least, error when an unknown attribute is used within `defmt_test::tests` because right now `cargo test`-ing this:
``` rust
#[ignore]
#[test]
fn loopback() { /* .. */ }
```
does run the test marked as `#[ignore]`

japaric, 2021-02-10:
I did `cargo expand` and it looks like we are discarding all the attributes that appear in the input code.

japaric, 2021-02-10:
This doesn't work either:
``` rust
#[cfg(ignore)]
#[test]
fn loopback() { /* .. */ }
```
in the sense that the test is compiled and executed when `--cfg ignore` is *not* passed to rustc.

jonas-schievink, 2021-02-11:
This now behaves as expected, do we want to close it, or leave it open to track `#[ignore]` support?

japaric, 2021-02-15:
I'll open a new issue about the `#[ignore]` attribute.

closed 2021-02-15 by japaric