knurling-rs/defmt #388: better support for software resets?

japaric, 2021-02-15:
this works: you get defmt logs from `probe-run`

``` rust
defmt::info!("Hello, world!");
cortex_m::asm::delay(10_000_000);
cortex_m::peripheral::SCB::sys_reset()
```

this does not, however: `probe-run` prints nothing

``` rust
cortex_m::asm::delay(10_000_000);
defmt::info!("Hello, world!");
cortex_m::peripheral::SCB::sys_reset()
```

if would be nice if both "Just Worked" but that would be complicated to implement because:

- the target zeroes the RTT control block on reset. That's what happening in the second snippet. The target writes to the RTT buffer, resets and then zeros the RTT control block (sets the RTT cursors back to zero). probe-run always observes the cursors as being zero
  - this may be workable by having the target *never* zero the RTT control block and instead have probe-run zero the control block once on startup. This way, after reset the target will continue to append data to the RTT buffer.

- a software reset can corrupt a defmt log frame (can it? I'm not 100% sure if that's the case with defmt-rtt because it disables interrupts while writing to the RTT buffer)
  - this may be fixable with some form of framing (e.g. COBS); probe-run would then be able to skip corrupted frames and move to the next one.

(framing would also be useful if we ever want to provide a transport layer that consists of a single-cursor circular buffer -- see second paragraph in https://github.com/knurling-rs/defmt/issues/249#issuecomment-726849253)

software resets are quite niche so this is definitively low priority

japaric, 2021-03-23:
in the example if I remove the delay I have run into a "failed to decode defmt data" errors in the probe-run

d3zd3z, 2023-11-06:
I encounter this when working on a bootloader that then chains to an application that also wants to use RTT for debugging.