knurling-rs/defmt #390: defmt-test: #[ignore] attribute support

japaric, 2021-02-15:
Support this

``` rust
// inside the `#[defmt_test::test]` module
#[ignore] 
#[test]
fn foo() {}
```

Tests marked as `ignore`-d should be compiled but not executed.
Ignored tests should be reported on the console output but should not be account for in the progress counter (e.g. `1/3`).

Urhengulas, 2021-03-16:
* Clarification question: The test should get compiled, but never executed. What will then get reported in the console?
* Also is compatibility with the `--include-ignored` flag desired? ([link](https://doc.rust-lang.org/reference/attributes/testing.html#the-ignore-attribute))

japaric, 2021-03-22:
> Clarification question: The test should get compiled, but never executed. What will then get reported in the console?

Ideally, something similar to what the built-in test harness reports: 
``` console
test tests::it_works ... ignored
```

> Also is compatibility with the --include-ignored flag desired?

we don't support "test" related flags in probe-run. That's blocked on [this RFC](https://github.com/knurling-rs/probe-run/issues/148)

japaric, 2022-01-14:
implemented in #618

closed 2022-01-14 by japaric