knurling-rs/defmt #402: defmt-test: README (Basic Usage section) should be updated

japaric, 2021-02-22:
Steps 1 and 2 refer to an old version of app-template that didn't use `defmt-test`. I think quite some time has passed so those two steps (and step 0) can be removed.

Urhengulas, 2021-03-16:
I am not sure if I understand the issue. Can you please rephrase it?

japaric, 2021-03-22:
> Steps 1 and 2 (and step 0)

[in this README file](https://github.com/knurling-rs/defmt/blob/55fcab61495b1651e2d157e79b50277f33c151ef/firmware/defmt-test/README.md#basic-usage)

should be removed

Urhengulas, 2021-03-23:
> > Steps 1 and 2 (and step 0)
> 
> [in this README file](https://github.com/knurling-rs/defmt/blob/55fcab61495b1651e2d157e79b50277f33c151ef/firmware/defmt-test/README.md#basic-usage)
> 
> should be removed

I got that it is about that file.

But I understood these steps to also be for people adding `defmt-test` to their existing project. But in that case they probably can refer to `app-template` and copy the stuff over from there.

commit 4bd7f502 (2021-03-29): Merge #436

436: `defmt-test`: Rm outdated setup steps r=jonas-schievink a=Urhengulas

Fixes #402
PR #436 (2021-03-29): `defmt-test`: Rm outdated setup steps
Fixes #402
closed 2021-03-29 by bors[bot]