knurling-rs/defmt #405: add `netlify.toml`

Lotterleben, 2021-02-22:
problem:

- we only have limited number of accounts to access Netlify and the admin who can reassign access is not part of the knurling project

Possible solution:

- configure `defmt` book deploys though a `netlify.toml` instead of the web UI https://docs.netlify.com/configure-builds/file-based-configuration/

possible bumps in the road:

- netlify docs are unclear on how to specify what the `production` branch is (since we have branches for diff releases this is the most important thing we'd like to be able to configure there so that's a must)

Lotterleben, 2021-02-22:
self-assigning for now; if someone gets impatient about this feel free to prod me or take it over

Urhengulas, 2021-05-05:
Solved by https://github.com/knurling-rs/defmt/pull/454

closed 2021-05-05 by Urhengulas