knurling-rs/defmt #407: defmt 0.2 does not compile after updating from 0.1

spcan, 2021-02-22:
I was using defmt v0.1.0 and run into a problem of infinite stack traces. To fix this, I updated to the latest defmt version (0.2).

Now, all builds fail with the next error:
```
warning: Linking globals named '__defmt_default_timestamp': symbol multiply defined!
error: failed to load bc of "defmt-ab550e4979f02e6f.defmt.7jti76t0-cgu.0.rcgu.o":
```

I've tried to compile with `cargo clean` as well as reverting to earlier versions (0.1.2 and 0.1.3). They all fail with this exact message, even with the same signature (ab550e4979f02e6f)

spcan, 2021-02-22:
Checking the output of `cargo build --release --verbose`, there are two link arguments passed into cargo:
`-L ~/target/thumbv7em-none-eabihf/release/build/defmt-00a44bb057f94192/out`
`-L ~/target/thumbv7em-none-eabihf/release/build/defmt-1eeeda97a72f4b5f/out`

jonas-schievink, 2021-02-22:
defmt 0.2 cannot be used together with defmt 0.1. Please make sure you aren't including both by inspecting your Cargo.lock or the output of `cargo tree`.

spcan, 2021-02-24:
That was it, I still had a small dependency logging with the old `defmt` crate. Thanks for the help.

closed 2021-02-24 by spcan