knurling-rs/defmt #409: defmt 0.2.0 - docs.rs build failed

japaric, 2021-02-24:
https://docs.rs/crate/defmt/0.2.0

jonas-schievink, 2021-02-24:
```
[INFO] [stderr]  Documenting defmt v0.2.0 (/opt/rustwide/workdir)
[INFO] [stderr] error[E0463]: can't find crate for `std`
[INFO] [stderr]   |
[INFO] [stderr]   = note: the `thumbv6m-none-eabi` target may not be installed
```
weird

jonas-schievink, 2021-02-24:
Ah, maybe we need to tell it to not pass `--all-features`?

Urhengulas, 2021-03-08:
This seems to be fixed, but it's not available on docs.rs. How can we update it there?

Urhengulas, 2021-03-08:
Reading the website it seems opening an issue is the way to request a rebuild (https://docs.rs/about/builds#failures-and-rebuilds). Should we do this?

jonas-schievink, 2021-03-08:
We can just publish 0.2.1 with this change included

Urhengulas, 2021-03-08:
Like now, or later?

jonas-schievink, 2021-03-08:
Feel free to do it now if you want. No other crates have to be published, just `defmt`, so it should be fairly straightforward compared to the 0.2.0 release.

Urhengulas, 2021-03-08:
Fixed by #413 and published by releasing `defmt v0.2.1` (#428).

closed 2021-03-08 by Urhengulas