knurling-rs/defmt #412: defmt-print: misleading error message on defmt version mismatch

japaric, 2021-02-24:
the error says to install **probe-run** instead of `defmt-print`

japaric, 2021-02-24:
``` console
$ defmt-print -e target/thumbv7em-none-eabihf/debug/hello
Error: defmt version mismatch: firmware is using 0.2, `probe-run` supports 0.1
suggestion: `cargo install` a different non-git version of `probe-run` that supports defmt 0.2
```

Urhengulas, 2021-05-12:
I was hoping to get the name of the binary from a cargo env variable during compile time, but none of `CARGO_PKG_NAME`, `CARGO_CRATE_NAME`, `CARGO_BIN_NAME`, `CARGO_PRIMARY_PACKAGE` seems to yield the name of the binary, but only the name of the library or nothing.

jonas-schievink, 2021-05-12:
We can probably solve this by using a strongly-typed error type in `defmt_decoder` that has a variant that indicates a version mismatch. Then the calling binary can print its own message.

Urhengulas, 2021-05-12:
> We can probably solve this by using a strongly-typed error type in `defmt_decoder` that has a variant that indicates a version mismatch. Then the calling binary can print its own message.

Yes, I also though about this, but then I tried to be smart 😬 😆