knurling-rs/defmt #416: defmt-test: support other items inside the `#[tests]` module

japaric, 2021-02-26:
This is currently rejected:

``` rust
#[defmt_test::tests]
mod tests {
    struct State {} //~ error

    #[init]
    fn setup() -> State {
        State {}
    }
}
```

But would be nice if structs, functions, etc. were accepted inside the module.

Right now you can work around the issue by putting helper structs and functions *outside* the `#[tests]` module and then importing them but this is cumbersome and quite different from how the standard `test` harness works.

``` rust
struct State {}

#[defmt_test::tests]
mod tests {
    use super::State;

    #[init]
    fn setup() -> State {
        State {}
    }
}
```

Sh3Rm4n, 2021-03-27:
I was conservative, when adding support for the use statement in https://github.com/knurling-rs/defmt/pull/294, but thought about use-cases like these as well. 

The question is, what items should **not** be accepted in the `#[tests]` module scope, when they are not also marked with `#[test]`, `#[init]` or something similar? 

When there aren't any, why not add support for every missing item at once?

Urhengulas, 2021-03-27:
> I was conservative, when adding support for the use statement in #294, but thought about use-cases like these as well.
>
> The question is, what items should **not** be accepted in the `#[tests]` [...]

Hi @Sh3Rm4n,

From my perspective we should be able to allow all items to enable an as smooth and familiar user experience as possible. Are you interested in doing these changes?

commit 697a8e80 (2021-04-29): Merge #441

441: Support more items in `#[tests]` module r=jonas-schievink a=Sh3Rm4n

This allows every item to be defined inside the `#[tests]` module,
e.g. `struct`s `const`s etc. except for function.

Closes #416
PR #441 (2021-04-29): Support more items in `#[tests]` module
This allows every item to be defined inside the `#[tests]` module,
e.g. `struct`s `const`s etc. except for function.

Closes #416
closed 2021-04-29 by bors[bot]