knurling-rs/defmt #429: const generics tracking issue

Urhengulas, 2021-03-09:
Since const generics will hit stable soon ([blog](https://blog.rust-lang.org/2021/02/26/const-generics-mvp-beta.html)), I started thinking about where we could use it and would like to track this in this issue.

**List**
- [ ] remove code duplication in `src/impls.rs`
  - [ ] arrays
  - [ ] ...
- [ ] ...

jonas-schievink, 2021-05-05:
We can't actually get rid of the duplication for arrays that easily, because every array length has a different format string, which is a literal.

However we could change how arrays work, and instead send the array length over the wire instead of encoding it in the format string. It would be somewhat less efficient but might be fine.

jonas-schievink, 2021-09-23:
We could actually keep the length in the format string for short arrays and send it over the wire for longer arrays. That way we could use const generics to implement `Format` for arrays of any size, without losing the efficiency of the current implementation.

closed 2021-09-24 by bors[bot]