knurling-rs/defmt #433: consider bundling the mdbook with API docs served on docs.rs

japaric, 2021-03-22:
This crate has en example [build script](https://github.com/japaric/cautious-octo-funicular/blob/v0.1.5/build.rs) that puts the mdbook in the `target/doc` directory.

[The rendered API docs](https://docs.rs/cautious-octo-funicular/0.1.5/cautious_octo_funicular/) contain an mdbook that looks like this

[image]

some extra CSS is need to hide or tweak the `docs.rs` nav bar to get a functional mdbook.

japaric, 2021-03-22:
^ we may want to do this so we get versioned mdbooks without having to set up multiple subdomains with netlify

Urhengulas, 2021-03-25:
Another option we started discussing is to migrate the book to `rustdoc`, similar how `egg` does it with their tutorial: https://docs.rs/egg/0.6.0/egg/tutorials/index.html