knurling-rs/defmt #449: panic-probe: Panic message isn't read from RTT before HardFault

mattico, 2021-04-14:
I'm using a fairly fast MCU (STM32H743XI @ 480MHz i-Cache and d-Cache enabled) with defmt-rtt and panic-probe on a J-Link probe using the WinUSB driver.

I never got a panic message when panics occurred - just a HardFault message. I assumed that the RTT polling didn't have time to read the panic message from the buffer so I [added a delay](https://github.com/mattico/defmt/commit/7e5756eff2aed15c965c0098bf6dc16b02ac9d7e) which worked about 70% of the time. Ideally the panic code could wait for the probe to read the message (if connected) before hard faulting - or maybe the probe could read the message even after hard faulting? Maybe there's a cache-related issue with the probe reading?

Urhengulas, 2021-04-15:
Hi @mattico, Seems like an interesting problem you discovered 😮

I'd assume we are able to wait for the probe in case of panic. Are you interested in implementing this?
Anyways, I will try get my hands on your kind of MCU to reproduce it :)

mattico, 2021-04-15:
I was just using defmt-test and never had an issue - presumably it doesn't do anything hardware specific like changing the core clock frequency or enabling caches so still probably one of those is the issue.

I could try to implement this. The one part I wasn't sure about is that `panic-probe` doesn't have any knowledge of the `Logger` implementation. I suppose we could add some (default impl?) functions to the `Logger` trait:

```rust
trait Logger {
    // TODO(defmt 0.3): Remove default impl
    fn is_connected(&self) -> bool {
        false // true?
    }
    fn write_buffer_len(&self) -> usize {
        0
    }
}
```

I'm very busy at the moment but if I get some free time or start having panics again I'll work on this.

jonas-schievink, 2021-04-15:
Are you using probe-run? It has this code in it, which looks like is trying to read more data after the MCU halts:

https://github.com/knurling-rs/probe-run/blob/df187189849c35cd00e50f2c2146d878c1636de1/src/main.rs#L507-L510

Maybe that isn't enough?

mattico, 2021-04-15:
Yes, I'm using probe-run. I'll look into that.