knurling-rs/defmt #450: defmt with rtic

davidlattimore, 2021-04-16:
I'm not sure whether to file this under rtic or defmt, but I'm having trouble using these together.

When I do, I get this:

```
    Finished dev [optimized + debuginfo] target(s) in 1.06s
     Running `probe-run --probe '0483:374b' --chip stm32g071 target/thumbv6m-none-eabi/debug/application`
  (HOST) INFO  flashing program (4.99 KiB)
  (HOST) INFO  success!
  (HOST) ERROR Max number of RTT attach retries exceeded.
Error: RTT control block not found in target memory.
- Make sure RTT is initialized on the target.
- Depending on the target, sleep modes can interfere with RTT.
```

That's when I do `cargo run` with an stm32g071 dev board attached. Relatively minimal source directory attached:

[defmt_and_rtic.tar.gz](https://github.com/knurling-rs/defmt/files/6322860/defmt_and_rtic.tar.gz)

If I comment out the rtic code and uncomment the regular `#[entry]`, then it works as expected.

Is it expected that these can't work together, or is there something I can do to make it work?

Thanks for defmt by the way, it's been awesome.

mattico, 2021-04-16:
If I remember correctly, I had this issue when I used `wfi` in my idle function rather than just looping.

Dirbaio, 2021-04-16:
Apply this workaround in `init`: https://github.com/probe-rs/probe-rs/issues/350#issuecomment-740550519

You'll probably have to adjust it for the `g0`.

davidlattimore, 2021-04-17:
Thank you. Both solutions worked!

I had no idle function, which presumably meant that rtic was putting the CPU to sleep. Adding an idle function with a loop made it work:

```rust
#[idle]
    fn idle(_ctx: idle::Context) -> ! {
        loop {}
    }
```

Alternatively, based of @Dirbaio's suggestion, adding the following to my init function fixed it.

```rust
ctx.device.RCC.ahbenr.modify(|_, w| w.dmaen().set_bit());
```

I now realise that probe-run mentioned in the error that "Depending on the target, sleep modes can interfere with RTT". I was thinking of sending a PR to add some documentation about how to make this work. Do you have any thoughts about where such documentation should belong? Alternatively, more detail could be added to the error - or perhaps a link to suggestions for how to resolve the problem.

Urhengulas, 2021-04-20:
Hi @davidlattimore, good to see that you were able to figure it out with the help of the community!

Such documentation would fit well into our "Troubleshooting" section in the REAMDE: https://github.com/knurling-rs/probe-run#troubleshooting
I am awaiting your PR! 😄

Urhengulas, 2021-05-05:
Solved by https://github.com/knurling-rs/probe-run/pull/183

closed 2021-05-05 by Urhengulas