knurling-rs/defmt #451: Display hint for unix timestamp to ISO

MathiasKoch, 2021-04-20:
It would be super nice with a display hint for formatting a unix milliseconds timestamp into an ISO8601 timestamp.

Eg: `1618910624804` -> `2021-04-20T09:23:44.804Z`

I would find this extremely useful for the log timestamps, as we are already sending `u64` sized timestamps over the wire, and could replace the current "uptime" with an actual RTC time.

jonas-schievink, 2021-04-20:
Seems like a good idea!

One question is whether we want this to LEB-encode the timestamp, which should be beneficial here. I think we might not be able to do this without a breaking change though.

RACERGUY71, 2021-05-21:
Yes for sher

japaric, 2021-07-02:
Adding a new display hint shouldn't be a breaking change so I'm going to remove this from the 0.3 milestone (i.e. this doesn't *block* the 0.3.0 release; it can be implemented anytime in the 0.3.x release series)